carla-simulator/carla #550: Measurements

mhusseinsh, 2018-07-02:
Hello,

I am running Carla on windows. I find from the documentation that there are a lot of measurements that are logged from Carla while running, however, I don't know where can I find these measurements. I searched for them in the whole directory, and I found nothing.

p-schulz, 2018-07-02:
Hi @mhusseinsh 
looking at other issues, you may find that the measurements are not dumped in any way by default. You have to fetch them using your client and dump them by yourself in the way you want it. Take a look at e.g. #446 for an example on dumping the location of the Ego vehicle into a .txt file.

marcgpuig, 2018-07-02:
Hey @mhusseinsh.
It is exactly like @p-schulz says. I recommend you to read the [docs](http://carla.readthedocs.io/en/latest/measurements/) about measurements.

mhusseinsh, 2018-07-18:
Okay thanks a lot :)

closed 2018-07-18 by mhusseinsh