carla-simulator/carla #551: Port Connection - Windows

mhusseinsh, 2018-07-02:
Hello, I am trying to run python client for CARLA on my windows machine and I receive an error with the port number

[image]

marcgpuig, 2018-07-02:
Did you modified the manual_control.py? Do you have the focus on the pygame window?

mhusseinsh, 2018-07-02:
no the manual_control is not modifed, and yes the focus is on the pygame window of the simulator

mhusseinsh, 2018-07-02:
Ok now I got it working, there are 2 pygame windows. One for the car, and one a view for the cameras, when I am on the car one, it doesn't drive, I have to be on the other one

marcgpuig, 2018-07-03:
Cool!

closed 2018-07-03 by marcgpuig

morningsky, 2019-05-10:
@mhusseinsh @marcgpuig  I met this same problem on latest 0.9.5 version， when I run ./CarlaUE4.sh ，it creates a camera pygame window, after that, I run manual_control.py then it creates a game window, but the keyboard always focuses on the camera windows. When I click the game window and press WASD, I can't control the car, but the camera keeps move. what should I do?

marcgpuig, 2019-05-20:
@morningsky Did you tried to change your keyboard to English?

morningsky, 2019-05-21:
@marcgpuig  Yes, it's ok when I changed it! Thanks!