carla-simulator/carla #553: Transform

Archiewyq, 2018-07-03:
hello,
i find the function **get_unreal_transform** in 

> ~/PythoClient/carla/sensor.py

but cannot understand the values in _`to_unreal_transform = Transform(Rotation(roll=-90, yaw=90), Scale(x=-1))`_, such as -90,-1

besides, i find in #56 : 

> Given a 2D point p2d = [u,v,1], your world point position P = [X,Y,Z] will be: 
> P = ( inv(K) * p2d ) * depth

In my understanding, it gives the Transform from Pixel coordinate to Camera coordinate 
so if i want to get results in  World  coordinate , i need the extrinsic matrix of my camera.
Is it correct?
thanks!

closed 2018-09-08 by stale[bot]

marcgpuig, 2018-09-12:
@Archiewyq Sorry for the late response.

Your thoughts are correct.
As you can see in the `point_cloud_example.py` on 0.8.X, we use this function to get the _camera to car_ transform:

[image]

Cheers!

zrfreya, 2018-12-20:
> @Archiewyq Sorry for the late response.
> 
> Your thoughts are correct.
> As you can see in the `point_cloud_example.py` on 0.8.X, we use this function to get the _camera to car_ transform:
> 
> [image]
> 
> Cheers!

@marcgpuig   Does this still work for carla 0.9.1 ?

marcgpuig, 2019-01-07:
Hi @zrfreya, sorry for the late response.

>  Does this still work for carla 0.9.1 ?

Yes, it does still the same :)

gxd1994, 2019-08-21:
> Hi @zrfreya, sorry for the late response.
> 
> > Does this still work for carla 0.9.1 ?
> 
> Yes, it does still the same :)

Does this still work for carla 0.9.5?

jnd77, 2020-07-21:
Hi.
This graph is great. Would it be possible to add one section about transformations and axis system in the documentation, and include this graph ? I think every newcomer to Carla spends some time initially to figure it out by going through github issues. Would be really helpful to have it in one place. :)

Thanks !

marcgpuig, 2020-08-18:
Hi @jnd77, I quickly made it a long time ago :grinning:
I'm sure that @sergi-e, our super docs guy, can do something better to add in our official docs.
Cheers!