carla-simulator/carla #554: Where is the documentation to use the cameras

Denis2054, 2018-07-03:
I can't find the documentation on the camera functions in Python.

Your Python examples work well. But I'd like to transform the manual example into an autonomous car as you suggest at one point. ("Can you do better?"). It's nice challenge...if there is documentation!

I know I can pick up the red channel from a camera but there is no function documentation. 

Thanks in advance for your answer!

marcgpuig, 2018-07-03:
Hello @Denis2054.

First of all thanks for your feedback!

Right now, our main teaching resource are the well documented python examples, where we try to use all carla module functions. The carla module is well documented too, but I understand is not convenient to go checking around other people's code.

I suppose you have checked the [Cameras and sensors](http://carla.readthedocs.io/en/latest/cameras_and_sensors/) documentation and so on, but you are right, this is not enough.

The problem is that this is not a good moment to do so, because we are changing all the server and client code... and in a few month everything will change (to better!). We'll try our best to improve the documentation for the next releases.

Really sorry for the inconveniences.

Denis2054, 2018-07-03:
Dear Marc,

Thanks a lot for your answer.

Your solution is worth waiting for the next version!

Best regards,

Denis

<https://www.packtpub.com/big-data-and-business-intelligence/artificial-intelligence-example>
*Denis Rothman*
*Artificial Intelligence expert, author, speaker*
<https://www.packtpub.com/big-data-and-business-intelligence/artificial-intelligence-example>
<https://www.packtpub.com/big-data-and-business-intelligence/artificial-intelligence-example>
Linkedin <https://www.linkedin.com/in/denis-rothman-0b034043>
Amazon Author Page <https://www.amazon.com/Denis-Rothman/e/B07DFRCJG8>
Packt Publishing Author Page
<https://www.packtpub.com/books/info/authors/denis-rothman>

marcgpuig, 2018-07-03:
Stay tuned for the next update :)

closed 2018-07-03 by marcgpuig