carla-simulator/carla #558: lidar data

mhusseinsh, 2018-07-05:
Hello

what is the difference between the 2 ply files generated, if I run `client_example.py -l` , and if I run `point_cloud_example.py `

nsubiron, 2018-07-10:
`client_example.py -l` saves the point cloud as generated by the Lidar. `point_cloud_exanple.py` generates a point cloud out of depth images.

mhusseinsh, 2018-07-10:
OK thanks
But if i compared the two measurements from the lidar directly and the generated ones from the depth images, is there a difference ?

Another question concerning the lidar, what is the limit of number of channels, points, and FOV ?

nsubiron, 2018-07-11:
The Lidar points are generated relative to the sensor, while the point cloud example does the conversion to world coordinates.

There is no limit in the number of channels and points, but the more you put, the slower will be the simulation. The FOV is limited between 90 and -90 degrees.

MKastuv, 2018-07-12:
Hi, I can run lidar using "python .\manual_control.py --lidar". But how can I save the lidar images ?

MKastuv, 2018-07-13:
Thanks @nsubiron, but looks like I cannot control the vehicle using .\client_example. Can I use .\manual_control.py and save the lidar data ?

nsubiron, 2018-07-14:
manual_control.py doesn't have that option, but you can pretty easily modify it. Right after
```py
measurements, sensor_data = self.client.read_data()
```
add the following
```py
for name, measurement in sensor_data.items():
    measurement.save_to_disk(filename)
```
or only the lidar data with
```py
sensor_data['Lidar32'].save_to_disk(filename)
```

MKastuv, 2018-07-16:
Thanks again @nsubiron, if I want to use the custom build map for the simulator, how should I do ?