carla-simulator/carla #560: manual control save images

mhusseinsh, 2018-07-05:
Hello,

I am trying to save images from the manual control.
I added the command lines which do save images

```
        # Print measurements every second.
        if self._timer.elapsed_seconds_since_lap() > 1.0:
            if self._city_name is not None:
                # Function to get car position on map.
                map_position = self._map.convert_to_pixel([
                    measurements.player_measurements.transform.location.x,
                    measurements.player_measurements.transform.location.y,
                    measurements.player_measurements.transform.location.z])
                # Function to get orientation of the road car is in.
                lane_orientation = self._map.get_lane_orientation([
                    measurements.player_measurements.transform.location.x,
                    measurements.player_measurements.transform.location.y,
                    measurements.player_measurements.transform.location.z])

                self._print_player_measurements_map(
                    measurements.player_measurements,
                    map_position,
                    lane_orientation)
            else:
                self._print_player_measurements(measurements.player_measurements)

            measurements, sensor_data = self.client.read_data()
            out_filename_format = '_out/episode_{:0>4d}/{:s}/{:0>6d}'
            step = self._timer.step
            episode = 0
            for name, measurement in sensor_data.items():
                filename = out_filename_format.format(episode, name, step)
                measurement.save_to_disk(filename)
```
However, I really notice that the difference between each consecutive image is too big (time wise). In client example, it captures at each frame, but not like here. Is there a way to edit this ?

nsubiron, 2018-07-10:
Hi @mhusseinsh,

If you put the code inside the
```python
# Print measurements every second.
if self._timer.elapsed_seconds_since_lap() > 1.0:
    # ...
```
then it's only called once per second.

closed 2018-07-18 by marcgpuig

mhusseinsh, 2018-09-11:
@nsubiron sorry for reopening the issue,
what you said will save the data every second so (1 fps), in the carla_client example, there is a higher frame rate

nsubiron, 2018-09-12:
Correct, just don't put your code inside 
```py
if self._timer.elapsed_seconds_since_lap() > 1.0:
```