carla-simulator/carla #561: Integrating OSM importer and spline generator

p-schulz, 2018-07-05:
Hi everyone,

along the possibility of generating roads and other structures from OSM data using something like CityEngine (see #540 for an very impressive example) I'm working on an approach of using Mike Frickers OSM Importer. I extended the importer (which is purely C++) to be available in blueprints (UFUNCTION-BlueprintCallable). This way it is possible to get the map data (e.g. roads) from a OSM file inside a blueprint and iterate over all available points. For now there is just a spline generated for each road, and for each roads node a spline point is added to its spline. I'm having a bit of trouble on adjusting the tangents of the spline points, but maybe this could be a good start on integrating OSM data in Unreal. The same way the roads are generated, one could easily iterate over the buildings and create a buildings that matches the original properties. 
Would there be any interest that I share / contribute my work?

[image]

marcgpuig, 2018-07-05:
Hello @p-schulz!

What an amazing work!
Surely there are people interested in what you are doing, also we are!
In my experience, OSM, that has a lot of information, do not provide an enough level of detail to generate realistic roads. We are thinking to use OpenDRIVE that can be also used to drive the cars in the enviroment.

Regardless of this, it will be really cool for the comunity to have integration with OSM, and I hope we could work together to achieve it :)

Feel free to make a PR anytime!

p-schulz, 2018-07-06:
Hi @marcgpuig!
great, I'm preparing the plugin and the blueprint for a PR :)

Using OpenDRIVE would indeed make more details available. Did you already have a workflow in mind? Such as using e.g. SUMO for converting basic OSM data to OpenDRIVE?

marcgpuig, 2018-07-06:
Hi @p-schulz!

Right now we are making a lot of changes in our code (you can see branch `carla-0.9`). I don't know if you want to wait until we release the new code... sorry about that :(

I want to take a look at [terminus](https://github.com/ekumenlabs/terminus), and see if it can help to the process of converting to OpenDRIVE, RNDF... so we could support as much as we can (if the license allow us). Also I will take a look at SUMO in the next weeks.

p-schulz, 2018-07-11:
A short update... I managed to handle the tangents. The tricky part now is setting up the intersections. I think I will wait for the new code, especially when advancing to 4.19, there might be some changes for the OSM Plugin to be made. Maybe I have improved the Plugin by then :)

Here are some shots of clean tangents:

[image]

[image]

[image]

This is actually a real place in germany (https://www.openstreetmap.org/search?query=ulm%20lehr#map=14/48.4332/9.9685).

I was also looking into another project: https://github.com/bneukom/UnrealLandscape
But this might be a bit overkill for the moment...

mohamednasrallah, 2018-07-16:
Hello,

This feature/plugin would be a great deal in my line of work.
When do you think this would be implemented in CarLa @marcgpuig 
Really excited to see it working
Thank you!

marcgpuig, 2018-07-17:
Hello @p-schulz @mohamednasrallah!
Sorry for the late response.

Your work is really impressive! It looks very good from above. Are the roads enough well generated to drive into? 

> I was also looking into another project: https://github.com/bneukom/UnrealLandscape

I've took a look at [UnrealLandscape](https://github.com/bneukom/UnrealLandscape) plugin and it looks amazing, thanks for sharing.

> When do you think this would be implemented in CarLa @marcgpuig

Right now I'm working on a library that, given certain OSM data, will try to solve some curves, intersections and junctions and generate a basic mesh. This library will be external to Unreal Engine, so it could be used everywere. You can take a look at the progress here #579.
However don't expect anything usable in a few months...

I think we could have better discusions on Discord, I'll make a proc-gen channel so everyone can give ideas and opinions. I encourage everyone to join.

Thanks for your interest!

marcgpuig, 2018-07-20:
Hi @p-schulz 
Click [here](https://discord.gg/8kqACuC) to join the Discord channel (you may need an account), and on the left, search for `# map-generation`.

Cheers!

closed 2018-09-25 by stale[bot]