carla-simulator/carla #562: bounding_box information and visualization

mhusseinsh, 2018-07-05:
Hello,

I can see that there are info about bounding_box should be provided in the measurments
my question is how can I have this info about the bounding_box ? and also, can the bounding_box be visualized either in realtime while running the simulation or in the saved images ?

p-schulz, 2018-07-05:
Searching for any issue regarding bounding box visualization, you could come across issue #314.

mhusseinsh, 2018-07-05:
thanks @p-schulz for your reply
I already did that and I checked the issue #314 as you said, I checked this [modified client ](https://gist.github.com/marcgpuig/8317fdb8ffd2a01f7156417d29d05857)that @marcgpuig  uploaded, and I downloaded it and run it, but it didn't draw any bounding boxes on any object, and since the issue was closed, so I asked again

Sorry for this misunderstanding

marcgpuig, 2018-07-06:
Hey @mhusseinsh.
Which version of CARLA are you using to test my modified client?

mhusseinsh, 2018-07-06:
@marcgpuig 0.8.4

marcgpuig, 2018-07-06:
@mhusseinsh 
As I said in #314:

> I modiefied a client so you can try it by yourself. It uses the **0.8.1** version of CARLA.

I've never tested it using the 0.8.4, probably this is why does not work fot you...
This was only a simple example so you can take a look and adapt the code.

closed 2018-07-18 by marcgpuig