carla-simulator/carla #563: Autopilot roadmaps for imported CityEngine models

abol-karimi, 2018-07-05:
Following the recommendation in [#540](https://github.com/carla-simulator/carla/issues/540) I imported a scene from CityEngine. It looks better than Infraworks models and the import is faster!

To setup the autopilot roadmap, my plan is to reimplement the method ACityMapGenerator::GenerateRoadMap(). However, there is a problem: A CityEngine's road segment is imported as a single mesh:

[image]

But
1. I need each lane to be a separate mesh so that the ray-casting in GenerateRoadMap() would work.
2. For curvy roads like in the above picture, I may even need to break down each lane into several (almost linear) segments.

Any advice on how to break down the meshes either in CityEngine or in Unreal?

Another approach I would imagine is to (manually, or even automatically?) add the road center curve (coming from OSM, ArcGIS, etc.) as an spline component to the road segment mesh. This way, after the mesh is determined by ray-casting, we may somehow compute the direction of the road from the spline. Again I'm not sure whether this is practical.

Best,
Abol

marcgpuig, 2018-07-06:
Hi @abol-karimi, nice job!

These all are the points we are investigating. As said in #540, we do not have an automatic way to generate the traffic paths for vehicles right now. I'm sorry but we can't help to much with this.

Using a spline to drive the cars is what we have in mind, but the idea is use OpenDRIVE because provides much more information. That's why we don't know which parts of our current code will be obsolete in a near future.

Also did you managed to get the roads metadata from CityEngine to Unreal? Not only the node positions but all the traffic information?

Regards.

closed 2018-09-11 by stale[bot]

hsienyan, 2018-11-30:
May I know which version of city engine you used？

XGodina, 2018-11-30:
Hi hsienyan, 

Nowaday we don't use city engine. Now we use RoadRunner. It is easier to use and we also export the OpenDrive. You can imported geometry and materials without problems.

This is the link and you can download a trial version. Only send a messatge. Tell them you're coming from Carla, they already know who we are.

https://www.vectorzero.io/?utm_source=so

hsienyan, 2018-11-30:
Thank @XGodina ,but CityEngine can directly download GIS data from google map.It looks nice to build city.

XGodina, 2018-11-30:
Well RoadRunner also you can use the GIS. But if you work with city engine, no problem. Wich software do you use to import geometries? Unreal ?

hsienyan, 2018-11-30:
I follow this
https://youtu.be/vMAVRnEWPEU

saicsk-deloitte, 2023-02-16:
Hi @hsienyan , After using above tutorial or using City Engine, Unreal. Have u succeeded in creation of new map for Carla simulator?