carla-simulator/carla #568: Vehicles and Map Dimensions

tayloryang722, 2018-07-06:
Hi CARLA team,

I have a few question about the model and maps dimensions.

1. Based on some previous issues, I knew that that the pixel density is 0.1643 m/pixel. #262 . Can I use this to calculate the distance traveled by measuring the pixels traveled?

2. Since I am interested in how far away the vehicle is from the intersection, could you give some information about the dimensions( Height, Width and Length) of both vehicles (Mustang and Model 3)? I am assuming that the GPS info of the vehicle is always located at the center of the vehicle according to your documentation.

3. How to measure the dimension(actual lenght) of both maps? Could you let me know what is size of maps in the unit of pixel so that I can calculate the length on my own?

I greatly appreciate every effort of your development team.

Sincerely,

marcgpuig, 2018-07-06:
> 1. Based on some previous issues, I knew that that the pixel density is 0.1643 m/pixel. #262 . Can I use this to calculate the distance traveled by measuring the pixels traveled?

I think it should do the job.
But you have the exact car positions each frame. So I think a more accurate solution should be adding each frame the distance between the old frame and the current frame.

> 2. Since I am interested in how far away the vehicle is from the intersection, could you give some information about the dimensions( Height, Width and Length) of both vehicles (Mustang and Model 3)? [...]

We provide the bounding box extent so you can calculate the bounding box of each vehicle. Take a look at these old issues: #64 #314.

> 3. How to measure the dimension(actual lenght) of both maps? Could you let me know what is size of maps in the unit of pixel so that I can calculate the length on my own?

You can open the Unreal Engine editor and measure it. @TheNihilisticRobot just made it for you (in m).
```
Town 1 - 410,68 x 344,26
Town 2 - 205,59 x 204,48
```

> 4 If I want to make the distance to intersection in town 1 and town 2 the same, should I use the same pixel and node density? Also, what is the dimensions of map town 2?

I do not understand your question :(

tayloryang722, 2018-07-06:
Hi marcgpoig,

Thank you for taking your time answering me questions.

I think that I have answers for most of my questions.

To clarify the last question, I want to calculate the distance from the vehicle's current position to the intersection. For example, my measurement of distance from current vehicle's position to the intersections in the unit of pixel might be different in two maps due to different dimensions of the map. But as long as I use the same pixel density, can I say if the measurement of distance in the unit of pixel  from current vehicle's position to the intersection is the same in both maps, the actual distance to the intersection will be the same?

Thanks,

marcgpuig, 2018-09-10:
@tayloryang722 Sorry for the late reply!
If I understand you correctly, you are right in your assumptions :)

closed 2018-11-16 by stale[bot]

ashutoshIITK, 2022-08-29:
@marcgpuig @TheNihilisticRobot Would it be possible to find the dimension of Town10 map? Thank you!