carla-simulator/carla #575: Changing camera position and orientation during simulation

sergiocaccamo, 2018-07-10:
Hi,

I'm trying to modify the position and rotation of a given camera in the scene without restarting the episode. Similar questions have been asked (#58 and #256) where you specify that it is not directly possible to do so with the current version. Since you are now working on a new server I wonder whether you plan to include this functionality. If not, do you have any suggestion for how to convert _request_new_episode() into _update_episode(settings). 

Thanks

marcgpuig, 2018-07-10:
Hi @sergiocaccamo.
It will be available in the next release and I recommend you to wait for it ;)

sergiocaccamo, 2018-07-10:
Wonderful! Release date? :D

marcgpuig, 2018-07-10:
@nsubiron did not seem very convinced while he was saying... _this month_

nsubiron, 2018-07-10:
This month, most probably :wink:

sergiocaccamo, 2018-07-11:
Great, thanks!

kylop, 2018-08-02:
Well done delivering in time ;)

marcgpuig, 2018-08-02:
As you said, the 0.9.0 is here! 
So I'm closing the issue :)

closed 2018-08-02 by marcgpuig