carla-simulator/carla #576: Adjust Pedestrian Animations

XGodina, 2018-07-12:
* [ ] Tweak the scale Pedestrians
* [ ] Fix the scale difference between Idle and walk animations
* [ ] Improve the cycle animations walk

YDSh510, 2019-05-21:
Hi @XGodina - How to add animation to pedestrian like making them walk.

1. Can it be done in Unreal itself or we need any other modeling tool to achieve the same and then import it in unreal?

2. I want to move the pedestrian hands up and down , is there any tutorial or link I can refer to achieve the same. 

Please guide as its really urgent.

XGodina, 2019-05-21:
Hi @YDSh510 

Here https://www.unrealengine.com/en-US/bing-search?keyword=animations. There is a lot of documentation and Tutorial about animations.

YDSh510, 2019-05-23:
@XGodina - Thanks for the link. Also wanted to know if there is any documentation w.r.t to pedestrian blueprint BP_Walker as I need to add logic for adding hand movement. So wanted to understand how Carla team have done for legs.

YDSh510, 2019-05-24:
Hi @XGodina - Any update w.r.t how to add hand movements in walker . ( Moving hands up and down)
As I don't see any option via CharacterMovement component.

YDSh510, 2019-05-27:
@XGodina - Need some urgent pointers to proceed further. Please support.

XGodina, 2019-05-27:
@YDSh510 I don't know what do you refer a w.r.t.  If you want to add new animation in pedestrian you have three ways.

1- Edit animation in Unreal - You can find a lot of tutorial on the internet
2- Edit in Maya and export to Unreal
3- We are working to control pedestrian by API. Will be available the next release.