carla-simulator/carla #579: Procedural road generation

marcgpuig, 2018-07-12:
# Introduction
As part of the **July 2018 Code Sprint**, we want to start generating some simple road cases.
We know this is a large and complex task, and some of you have already started doing it (with good results). So we want to unify all these work and coordinate all the efforts to achieve a good procedural road generatio.

I was searching for C++ MIT geometric libraries to help with the development and found [Wykobi](https://github.com/ArashPartow/wykobi). If anyone knows other C++ libraries feel free to comment below ;)

This feature can be splitted in the following tasks:
- [ ] Setup an external library for CARLA that will allow a modular development for the community
- [ ] Use an OpenGL pltform to develop the initial parts of the library
- [ ] Resolve simple curves
- [ ] Resolve simpe intersections
- [ ] Road mesh generation
- [ ] Road lines mesh generation
- [ ] Texture generation
- [ ] Driving waypoint generation
- [ ] Integrate the library to UE4 and use OSM information to generate the roads.

# Use Example

yiakwy, 2018-07-18:
I am doing it in a different way. I strictly followed OpenDrive first, then I used made my own HD map format for map data structure. My map consists of three layers: Topological Layer, Geometry Layer, and Transportation Layer. By applying L-system alike algorithms upon my HDMap I can produce faked with totally under my control:

Here is why? Engineers find that AV tests are extremely hard and they what find 5% traffic cases which impact the algorithms a lot. I made a parametric generation control panel so that I can control the curvatures, number, and type of junctions and traffic flows.

I think carla put too much effort into rendering and manually modification of a map.

mohamednasrallah, 2018-07-18:
Thank you guys for working on this. This makes CarLa a lot more valuable for our work.

I've been trying to solve this quickly with my own work.

Currently for my AV team, I'm working on importing road xyz coordinates per spine node instance.
This import comes from a CSV file with XYZ columns.

### **The mechanism is w.i.p, but should be as so:**

1) Import CSV coordinates into CarLa/Unreal
2) set spline with road mesh to start from first xyz
3) set next spline node to next coordinate from file
4) conitnue until end of file

### **Issue**

 - We do not have tangent line information, so I'm assuming the tangent lines will take default values just like how carla has them set up.... or the tangent will depend on prev/next nodes

My work is less on importing actual maps straight from the source, but just to have it take in any map coordinates we have to generate high res road curvature just from multiple spline node isntancing.

I'm having slight issues with it, but anyone's input would be nice :) initially we're doing this because Apollo's opendrive format is annoying to convert into something unreal can use. So this should be an easy method to do it quickly. (We're on a tight release schedule for simulation)

ostromann, 2019-05-14:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

Anyone still working on this?

germanros1987, 2020-03-19:
Before closing this issue, it is worth highlighting the amazing work that @marcgpuig has done in the generation of mesh from OpenDRIVE. If you are reading this and don't know about the feature, please check out http://carla.org/2020/03/09/release-0.9.8/

closed 2020-03-19 by germanros1987

yiakwy, 2020-04-01:
I am glad to hear about that. Congratulations! Instead of OpenDRIVE, since
I already got full scene of HDMap arch and implementation, I also recommend
you refer to NDS in the persistent layer of an HDMap. I also
maintain patents about generating curves from catmul , bspline to clothoids
which are crucial to L-system a like generation algorithm.