carla-simulator/carla #580: Question about average distance between collision metrics

jackspp, 2018-07-13:
Hi, 
I am using CARLA 0.8.2 now. When testing in benchmark, I can get results like 

> Avg. Kilometers driven **before** a collision to a PEDESTRIAN
However, in CORL paper, as for the second table, the metrics are like

> Average distance (in kilometers) traveled **between** two infractions

I am wondering if this is a description problem since they are the same,
or Extra codes need to be written for calculating distance **between** two infractions?

mvpcom, 2018-07-14:
Hey @jackspp, you can find a code for automatically reproducing the same tables as [Carla Paper](http://carla.org//2018/04/23/release-0.8.2/#can-you-beat-our-agents-) in my [github repository](https://github.com/mvpcom/carlaILTrainer). Please check it out and feel free to let me know about any problem or suggestion.

felipecode, 2018-07-17:
This description was maybe misleading.
We take the total number of kilometers and divide by the total number of infractions.
We also have a way to reproduce the carla paper tables. Please check the version 0.8.2
blog post. The driving benchmark module:

http://carla.org/2018/04/23/release-0.8.2/

closed 2018-09-22 by stale[bot]