carla-simulator/carla #581: understand lidar ray-cast

mhusseinsh, 2018-07-13:
Hello,
I am kind of new a little bit to lidar data and usage
I understand the data of the lidar from points, channels, FOV and everything
But I want to understand what is the ray-cast ?

luigui2906, 2018-07-13:
Hi @mhusseinsh,

I think the ray-cast operation consists in following the ray from the origin (the sensor) to the intersection with a surface which in this case would represent the coordinates of the impacted point. I guess this is what is done in order to obtain the point cloud data in the simulator.

Hope this helps.

mhusseinsh, 2018-07-16:
Thanks a lot :)

closed 2018-07-16 by mhusseinsh

hamza422, 2021-01-20:
@@mhusseinsh @marcgpuig  I want to ask a question-related ray cast. As they explain in Unreal's article to highlight the object where the ray hits. I want to make it work in Carla. Can Lidar sensor do raycasting in a pointing direction? like a ray?
Would you help me with this?