carla-simulator/carla #582: error occured when enable lidar publication for carla_ros_bridge

lxtttt, 2018-07-13:
Hi,
I am trying to enable lidar publication and uncomment those lines in setting.yaml but i got error.
Error message:
Traceback (most recent call last):
  File "/home/ubuntu/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 34, in <module>
    main()
  File "/home/ubuntu/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 30, in main
    carla_ros_bridge.run()
  File "/home/ubuntu/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 156, in run
    self.sensors[name].process_sensor_data(data, self.cur_time)
  File "/home/ubuntu/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/sensors.py", line 59, in process_sensor_data
    self._compute_transform(data, cur_time)
  File "/home/ubuntu/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/sensors.py", line 107, in _compute_transform
    t.header.stamp = self.cur_time
AttributeError: 'LidarHandler' object has no attribute 'cur_time'
[INFO] [1531471478.126028]: Shutdown requested
[carla_ros_bridge-2] process has died [pid 15009, exit code 1, cmd /home/ubuntu/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/client.py __name:=carla_ros_bridge __log:=/home/ubuntu/.ros/log/f74e5d78-8678-11e8-95c8-d050992713c6/carla_ros_bridge-2.log].
log file: /home/ubuntu/.ros/log/f74e5d78-8678-11e8-95c8-d050992713c6/carla_ros_bridge-2*.log

Everything runs well when i didn't do the uncomment
Can anyone help me? Thanks a lot!

danil-tolkachev, 2018-07-13:
Hey @lxtttt. Try to use latest code from the master branch. It was already fixed.

marcgpuig, 2018-07-16:
Hi @lxtttt.
Close the issue if @danil-tolkachev's answer works for you :)

closed 2018-07-16 by lxtttt

wolfsubak, 2018-08-22:
The fix is not available on the 0.8.4 branch and ros is not supported yet on 0.9.0 is there a way to solve this issue ?

danil-tolkachev, 2018-08-22:
@wolfsubak 
Latest binary files from version 0.8.4 and python code from the master branch (`PythonClient` and `carla_ros_bridge`) should work.

@nsubiron 
Could you clarify the future of 0.8.x branch? Are there any plans to 0.8.5 version? Will the stable branch always point to 0.8.2 or will it shift to later versions?

danil-tolkachev, 2018-08-22:
@wolfsubak
You are using the old ros-bridge. You should replace it with the code from the master branch.

danil-tolkachev, 2018-08-22:
@nsubiron 
Thanks for the detailed answer. I'll think about possible features in the API.

wolfsubak, 2018-08-23:
@danil-tolkachev oh perfect, understood. It was a little bit confusing between using stable carla and master ros bridge.
It works perfectly on 83070b8c8f9cde1257b4ed6c467e7a2ee3c2f14d but not on master by the way, an error occured since last commit.

danil-tolkachev, 2018-08-30:
Hi @chengjinjack
I'm not sure. Sometimes test runs with warnings on my machine. But I've never seen your error. Have you checked the ros-bridge with bundled launch files?

And the traditional question: 
Did you check that you are using `PythonClient` and `carla_ros_bridge` from master branch with binary files from release 0.8.4?

chengjinjack, 2018-09-01:
@danil-tolkachev
Thanks.The issue has been solved!