carla-simulator/carla #583: Noise on Point Cloud Data

luigui2906, 2018-07-13:
Hi,

I would like to know if the Point Cloud data obtained comes already with some simulated noise or instead it is perfect data.  I think is the second one, then I should add the noise by myself if I need it. I just want to confirm.

Thanks for your help and the great work you are doing with the simulator.

marcgpuig, 2018-07-16:
Hi @luigui2906!
Your assumptions are correct! 
You can add a _ditsortion_ by car movement running the simulator using high `fps`.
For instance:
```
-benchmark -fps=60
```
Also modifying the lidar's frequency and the angle of information gathering.

Cheers!

luigui2906, 2018-07-16:
Thanks @marcgpuig ..!

closed 2018-07-16 by luigui2906

Tonyfanto, 2019-06-29:
Just to confirm .So by increasing the fps from 10 to 60 does it add noise to the lidar data ?Also how the lidar's frequency and angle of information should be?