carla-simulator/carla #584: Carla Simulator gets stuck

kk2491, 2018-07-14:
Hi Guys,

Firstly Thanks for this wonderful work.

I recently started working on Carla simulator. But during initial stages itself I am having some basic issues.

My system specification is given below:
Ubuntu 16.04 LTS
RAM - 4 GB
CPU - 2 GHz
Nvidia graphic card - 2 GB NVIDIA GEFORCE (I am not sure how to make use of GPUs in case of Carla Simualto

Whenever I run the simulator there is a lag and the video gets stuck.

Could anybody please check and let me know if the above system specification is sufficient to run the simulation.

Thank you,
KK

marcgpuig, 2018-07-16:
Hi @kk2491,
Which NVIDIA GEFORCE do you have?
Also make sure you are not running the simulator on "server mode", so it gets stuck because is waiting for a client to connect!

kk2491, 2018-07-16:
@marcgpuig .. Thanks for the response.. I will check the Nvidia version and let you know ... 
I am not running the simulation in "server mode". 
Just to update it works better if I run in Windows 10, but in Ubuntu it hardly moves.

marcgpuig, 2018-07-16:
@kk2491 In Windows we experiment a better performance too, so maybe is your hardware :(

kk2491, 2018-07-16:
Windows I see better performance. but since Carla is more friendly in Ubuntu I badly want to try in Ubuntu rather than Windows.

marcgpuig, 2018-07-16:
You need a good hardware to run CARLA, that is true... sorry about that!

kk2491, 2018-07-17:
@marcgpuig Thanks for the response ... Please let me know if you have any minimum hardware specification to run Carla ..

marcgpuig, 2018-07-17:
Hey @kk2491.
We do not have such a specs, I'm sorry...

kk2491, 2018-08-03:
Thanks @marcgpuig .. I think we are good to close this.

closed 2018-08-03 by kk2491

Jayveer722, 2022-03-25:
System requirements

x64 system. The simulator should run in any 64 bits Windows system.

165 GB disk space. CARLA itself will take around 32 GB and the related major software installations (including Unreal Engine) will take around 133 GB.

An adequate GPU. CARLA aims for realistic simulations, so the server needs at least a 6 GB GPU although 8 GB is recommended. A dedicated GPU is highly recommended for machine learning.

Two TCP ports and good internet connection. 2000 and 2001 by default. Make sure that these ports are not blocked by firewalls or any other applications.

In practice the system requirements are driven by how you're using it. If you've got a sparse environment with zero cameras the requirements are quite low. You try loading up town10 with multiple cameras on 4gb of gpu mem and unreal will laugh and immediately run out of gpu memory.