carla-simulator/carla #587: How to define a start position in example_python.py

Denis2054, 2018-07-16:
I'm beginning to run  CNNs on the frames and include it in example_python.py

How can I define a start position in Town01.

I can't find a way to make it start from location 112,110,38, for example.

marcgpuig, 2018-07-16:
Hi @Denis2054.

We currently do not support this feature right now (until next release).
You have a list of predefined spawn points. If you want to add a new one, you must add it from the Unreal Editor.

Denis2054, 2018-07-16:
Hi,

OK. Thanks for your answer. No problem.

Best regards,

Denis

-- 

Denis Rothman
<https://www.packtpub.com/big-data-and-business-intelligence/artificial-intelligence-example>
Artificial Intelligence expert, Author, Speaker
<https://www.packtpub.com/big-data-and-business-intelligence/artificial-intelligence-example>
Linkedin <https://www.linkedin.com/in/denis-rothman-0b034043>
Amazon Author Page <https://www.amazon.com/Denis-Rothman/e/B07DFRCJG8>
Packt <https://www.packtpub.com/books/info/authors/denis-rothman>
Publishing Author Page
<https://www.packtpub.com/books/info/authors/denis-rothman>

marcgpuig, 2018-07-16:
Regards :)

closed 2018-07-16 by marcgpuig