carla-simulator/carla #589: Signal 11 caught

Ballester, 2018-07-16:
Specs:
pre-compiled CARLA 0.8.3
Ubuntu 16.04.3

When starting CARLA in a server through SSH, I get the following error:

> Increasing per-process limit of core file size to infinity.
> Signal 11 caught.
> Malloc Size=131076 LargeMemoryPoolOffset=131092 
> CommonLinuxCrashHandler: Signal=11
> Malloc Size=65535 LargeMemoryPoolOffset=196655 
> LogPlatformFile: Not using cached read wrapper
> LogInit: Display: RandInit(720732688) SRandInit(720732688).
> LogTaskGraph: Started task graph with 4 named threads and 9 total threads with 1 sets of task threads.
> LogLinux: Error: appError called: Assertion failed: IsValidIndex(Index) [File:Runtime/Core/Public/Containers/UnrealString.h] [Line: 218] 
> String index out of bounds: Index 1 from a string with a length of 1
> 
> LogLinux: === Critical error: ===
> 
> 
> LogLinux: Assertion failed: IsValidIndex(Index) [File:Runtime/Core/Public/Containers/UnrealString.h] [Line: 218] 
> String index out of bounds: Index 1 from a string with a length of 1
> 
> 
> 
> LogLinux: FLinuxPlatformMisc::RequestExit(1)
> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> Segmentation fault (core dumped)

This error happens when running in the standard way and when following the steps in the headless tutorial (https://carla.readthedocs.io/en/latest/carla_headless/)

I tried to run with docker but also with no luck.
Any clue on why this error is raising?

marcgpuig, 2018-07-17:
Hi @Ballester!
Can you reproduce this error on version 0.8.4?
Also make sure you are not running the SSH with the `-x` argument.

Ballester, 2018-07-17:
Yes, the error persists in 0.8.4
I tried both with and without -X in ssh with no changes.

I wonder if this has anything to do with the X server configuration in my server. Is there a way to disable all graphics and such before running CARLA?

tomchen1000, 2018-07-30:
You can run with env variable SDL_VIDEODRIVER=offscreen, e.g.
```
SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -carla-server -carla-no-hud
```

andr0idsensei, 2018-09-27:
I have the same issue when trying to run CARLA on a server with 4 gpu's, with no display. I am using CARLA 0.9.0 and Ubuntu 16.04 with Nvidia driver version 390.30. It happens even when trying to run CARLA as stated above.

materdd, 2018-10-09:
Same here!!  I am using CARLA 0.9.0 and Ubuntu 16.04 with Nvidia driver version 390.77.

nsubiron, 2018-10-11:
Maybe related to #757?

materdd, 2018-11-06:
I solved this problem. I changed nvidia driver from 390.77 to 390.87. And only worked for calra 0.8.3, not for 0.8.4, 0.9.0. I think this is emergency treatment, but please try that.

micmarty, 2018-12-07:
I have the same problem as OP.
```
Ubuntu 16.04 Server
Carla 0.9.1 release
NVIDIA 410.73
CUDA 10.0
TITAN X +  GTX 750i
```

Using virtal display:
`xvfb-run --auto-servernum -s "-screen 0 1024x768x24" bash`
then
`./Carla -opengl3` <- because "my" machine has no OpenGL from NVIDIA drivers
INB4 I'm not allowed to change NVIDIA driver version.

[image]

closed 2019-02-13 by stale[bot]