carla-simulator/carla #590: Crash on 0.8.4

felipecode, 2018-07-16:
####Specs:
I ran CARLA 0.8.4
Connected a client with python3
It crashed in less than 72 hours. 
The number of pedestrians is 200.
Number of vehicles 0
The Bikes were not disabled but there were no vehicles
I am attaching the logs

[CarlaUE4.log](https://github.com/carla-simulator/carla/files/2198645/CarlaUE4.log)

nsubiron, 2018-07-17:
```
[2018.07.14-02.07.23:428][162]LogSpawn: Warning: SpawnActor failed because of collision at the spawn location [X=-368.000 Y=25136.000 Z=127.000] for [YamahaBp_C]
[2018.07.14-02.07.23:464][162]LogCarla: Starting the timer to spawn the other -1 vehicles, one per 3.000000 seconds
```
Nice, so it first spawns 1 vehicle then spawns -1 vehicle, seems optimal to me :smile:

This seems like an issue of the vehicle spawner that it actually tried to spawn a motorbike. I'll take a look.

nsubiron, 2018-07-17:
I'm not able to reproduce. I also noticed that in the last episode you actually requested 20 vehicles.

felipecode, 2018-07-17:
I don't know how that would happen. I don't change the configuration file.
I  noticed that the last episode was run with something like a default configuration.
20 vehicles, 30 pedestrians, weather 1

Anyway
I tried again also disabling the bikes. I haven't seen the crash in 24 hours. 
I am still having some weird connection problems. But this could be on my side.
I will retry with version 0.8.2

Thanks.

felipecode, 2018-07-19:
So, conclusion.
You should set the flag:
`conditions.set(DisableTwoWheeledVehicles=True)`
Without this flag, it seems CARLA can crash even without vehicles..

closed 2018-07-19 by felipecode

nsubiron, 2018-07-19:
Still worth noticing that there is a bug in the current vehicle spawner. We are not fixing it for now as the vehicle spawner is most probably going to disappear in the 0.9.0 release.