carla-simulator/carla #595: Is there any way to change position when player vehicle is running?

mznw, 2018-07-18:
Is there any way to change position when player vehicle is running? Just changing the position and heading，and speed remains the same

marcgpuig, 2018-07-18:
Hi @mznw.

This is not currently possible right now, but it will be possible with the changes in the nex release :)
Take a look at #439.

marcgpuig, 2018-09-17:
@mznw This is currently possible on 0.9.0!

closed 2018-09-17 by marcgpuig

SuperShinyEyes, 2021-09-30:
> @mznw This is currently possible on 0.9.0!

Hi Marc, thanks for the great work. It'd be helpful for us if you could point a specific PR which resolved this issue. :)

Himverma2012, 2022-09-19:
Hi, I want to read the location of vehicle CG from a file and change the vehicle location accordingly. I also want to change the rotation of the vehicle at each timestamp while carla simulation is running. How to do that? get_transform() is there but there's no 'set_transform()' function. What's the workaround?