carla-simulator/carla #596: Epic: Logging and Playback functionality

iFuSiiOnzZ, 2018-07-18:
### Abstract
At a given point in time the user can start to record all the events that happen in the world so he can replay it back later. While is watching the replay the user can forward or backward and at any point during the replay he may want to stop it keeping the world state giving the possibility to the user to apply new inputs.

### Types of replay systems
There are two ways of doing a replay system:
1. storing user event
2. storing game state

Taking the first approach can make the system maybe a bit more easy to implement because we will only log the user inputs and AI actions, which also make the generated files no to be so heavy, but if in stead we need to control the replay at different speeds, go back or forward, etc saving the game state make it more easier. So taking into account the requirements the best option is storing the game state.

### What to store
One of the problem of storing the game state is the amount of memory that could use because at each frame a snapshot of the state of the application is done. So it will be necessary to find a way to reduce the amount of information stored per frame. All the static object will be stored only once, of the dynamic objects will store information only when it change and if is not changing will make reference to the last frame that the object changed.

Per each one of the dynamic objects of the scene we will have to store the next information:
- bounding box
- direction vector
- rotation, scale, position
- object identifier
- acceleration
- timestamp
- velocity
- steering
- other events (weather,  map change, sensors, etc)
- AI  (respawn, etc)

### Storing file format
There are two ways to store the information:
1. binary file
2. text file

Using a binary file makes it easier to save/read the information to/from disk as the information is the same that is in memory. On the other hand it makes it harder to edit it as is not readable for human, so it will be necessary some extra tools to allow users to see/edit the information. The other option is to store it as a text file, which make the information easy to read/edit for human but it will need some transformations to save/access it. As we don't have a clear definition about if the user will like or not to be able to edit the file will take the text file approach, and practical reasons the information it will be stored in a json file.

analog-cbarber, 2018-07-18:
If you are going to save this much information, you will really want a compact binary format, and might event want to consider using fast compression. It can make a huge performance difference. Human readable is fine for config files and logging but really costs you for high volume data.

iFuSiiOnzZ, 2018-07-18:
Hi @analog-cbarber 

You are right about the file size, performance, etc and it would also be easier to deal with the binary than the text file, but we are still waiting to see which option is the best for the researchers, for that reason we choose the text file for now.

Greeting

closed 2019-03-29 by stale[bot]