carla-simulator/carla #603: Road from SlpineMesh Repeater

PicheritB, 2018-07-18:
Hi all,

I'm currently testing Carla environment for my work, and I'm stuck on map customisation problems.
I would like to understand how to, after having created a curved road from Spline Mesh Repeater, have that newly created road considered as a road by the vehicles. For the moment the vehicles won't follow the roads. 

An example from an empty map (according to [map_customisation](http://carla.readthedocs.io/en/latest/map_customization/) documentation). My actions were: 
- Create my road with Spline Mesh Repeater. This road is a curved one that shall join two automatically generated roads (with CarlaMapGenerator)
- Set a Route Planner for the vehicle to turn on my new road (as explained in documentation).
- Bring the SplineMeshRepeater under the CarlaMapGenerator in the hierarchy of "World Outliner"
- Trigger "Road Map Generation" in CarlaMapGenerator

When starting the simulation, the vehicles do follow the Road Planner, but when arriving at the end of the Road Planner the vehicles are lost (i.e. they don't keep on following the curved road). See image below. Is there something I missed ? I see from [map_customisation](http://carla.readthedocs.io/en/latest/map_customization/) image that the straight roads are filed with green arrows telling the vehicles the good direction… How can I set this on my curved road ? 

I know you're planning to improve custom map generation on future releases ([#579](https://github.com/carla-simulator/carla/issues/579)) so the answer to my question might be "coming soon", but I have the felling I missed something.

Thanks your help, and well done for all the work on that project, it is quite impressive ! Looking forward to next evolutions !

[image]

TheNihilisticRobot, 2018-07-18:
Hi @PicheritB 

That's a nice approach as to how to build roads outside our RoadMapGenerator. Unfortunatelly the issues you are having are to be expected.

All roads generated withing the RoadMapGenerator are who give the vehicles the information they need to keep going forward. Outside of these roads the vehicles need to be manually guided by you, otherwise they'll wander trying to get to the road again (they start going forward and steering to the right constantly). 

You could keep adding points to the RoutePlanner until you finish guiding them trough the added road, then the vehicles would get back to the common road and keep driving by themselves.

As you've told we are fully renewing our road generation system so you won't have to worry about this kind of thing soon.

PicheritB, 2018-07-20:
Hi @TheNihilisticRobot,

Thank you for the quick answer. I understand this is not the way to do integrate a road to RoadMapGenerator.
In current CARLA version, is there a way to do so (for straight/curved road) (other than using the CarlaMapGenerator, with which you can't draw the roads as you like)? 
Otherwise, I will wait for the next version :)

Thank you

TheNihilisticRobot, 2018-07-24:
I'm sorry but we have not yet  implemented a system to create roads outside the grid that CarlaMapGenerator generates. We are working on a better system for the future though!

closed 2018-10-06 by stale[bot]