carla-simulator/carla #604: Measurments: player and non-player -- transform (x,y,z) location coordinates

mhusseinsh, 2018-07-18:
The transform of the agent is giving a location coordinates of x,y,z in meters.
This distance is respect to what ? so x,y,z from which point ?

also, when I tried to check the measurements of non-player agents, i believe it printed all the agents inside the simulation, not in the current scene.

Is there a way to get the measurements in the current scene (frame) that I am only in ?

mhusseinsh, 2018-07-18:
Simply, I am trying to collect a dataset from the simulator to run an algorithm of objects detection.
So I want the measurments (location) of each object detected in the scene

marcgpuig, 2018-07-18:
> This distance is respect to what?

The `(0,0,0)` of the Unreal's scene. This point is arbitrary. This was already answered here #355.

> Is there a way to get the measurements in the current scene (frame) that I am only in?

Not currently, this will be implemented with the new networking architecture.
But you can calculate it in your python client and discard the non-player measurements that you don't want.

mhusseinsh, 2018-07-19:
hello @marcgpuig 
thanks for your quick reply, so does this mean also that the transform of the non-player agents is with respect to the same point, or for these agents, it is respect to the player agent (the vehicle)

Secondly, any tips and tricks of how to calculate in the python client to save only the non-player measurements available in the scene ?

Last question, is there a way to have the trajectory of the vehicle since the start of the episode till it finishes ?

marcgpuig, 2018-07-19:
Hi @mhusseinsh.

> so does this mean also that the transform of the non-player agents is with respect to the same point, or for these agents, it is respect to the player agent (the vehicle)

Yes

> any tips and tricks of how to calculate in the python client to save only the non-player measurements available in the scene?

I've don't investigated it, but it shouldn't be complicated. I'm sure you can find a lot of information about this topic.

> Last question, is there a way to have the trajectory of the vehicle since the start of the episode till it finishes?

It is up to you do so. But you can store each position on a list, right?

Cheers

closed 2018-09-17 by marcgpuig