carla-simulator/carla #611: Python code to start the Python Client examples at a fixed location

Denis2054, 2018-07-20:
Does anyone have a line of Python code example to include in the Python client examples to start the SDC at a given existing location in the view_location.php list?

I'm trying to analyze one itinerary and it keeps starting at random locations.

Thanks in advance!

marcgpuig, 2018-07-20:
Hi @Denis2054 
If you take a look at `client_example.py` you will find these lines:

https://github.com/carla-simulator/carla/blob/860dbe908e7a81232bdba6853943707f6b0102c2/PythonClient/client_example.py#L99-L107

Here you can see that `client.start_episode(...)` expects an integer ID between 0 and `len(scene.player_start_spots) - 1`

I recomend you to take a look at `client_example.py` to familiarize yourself with Carla module.

Cheers.

Denis2054, 2018-07-20:
> 860dbe908e7a81232bdba6853943707f6b0102c2/PythonClient/
> client_example.py#L99-L107

OK. Thanks a lot!!!

<https://www.packtpub.com/big-data-and-business-intelligence/artificial-intelligence-example>
*Denis Rothman*
*Artificial Intelligence expert, Author, Speaker*
<https://www.packtpub.com/big-data-and-business-intelligence/artificial-intelligence-example>
<https://www.packtpub.com/big-data-and-business-intelligence/artificial-intelligence-example>
Linkedin <https://www.linkedin.com/in/denis-rothman-0b034043>
Amazon Author Page <https://www.amazon.com/Denis-Rothman/e/B07DFRCJG8>
Packt Publishing Author Page
<https://www.packtpub.com/books/info/authors/denis-rothman>

closed 2018-09-19 by marcgpuig