carla-simulator/carla #617: Engine crash handling finished; re-raising signal 11 for the default handler.

melfm, 2018-07-21:
Having generated new maps, I'm trying to create a binary version of CARLA, following the build instructions and packaging using the provided `Package.sh` was successful however when I try to run the server executable, I get the following error:

```
./CarlaUE4.sh 
Increasing per-process limit of core file size to infinity.
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
Malloc Size=131076 LargeMemoryPoolOffset=327748 
Malloc Size=131160 LargeMemoryPoolOffset=458936 
Malloc Size=131160 LargeMemoryPoolOffset=590120 
Malloc Size=131160 LargeMemoryPoolOffset=721304 
Malloc Size=151763 LargeMemoryPoolOffset=873091 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

When I run with gdb:

```
gdb CarlaUE4 :

Thread 1 "CarlaUE4" received signal SIGTRAP, Trace/breakpoint trap.
FLinuxPlatformMisc::DebugBreak ()
    at ~/UnrealEngine_4.18/Engine/Source/Runtime/Core/Private/Linux/LinuxPlatformMisc.cpp:765
```

- Computer setting:
   - cpu: i7-6850K CPU
   - gpu: TITAN X (Pascal)
   - NVIDIA driver version : 390.77

Any ideas on how to investigate this issue further?

melfm, 2018-07-22:
Actually I realized that I was making a mistake and trying to run `UnrealEngine_4.18/carla/Unreal/CarlaUE4/Binaries/Linux` executable which isn't intended to be run as a standalone. And the actual packaged up dist will be under `UnrealEngine_4.18/carla/Dist/0.8.4-24-xxxxx/LinuxNoEditor` and this works as expected.

closed 2018-07-22 by melfm

chankim, 2018-09-07:
@melfm Thanks for the info. I was in the same situation. But do you know what  this 'LinuxNoEditor'  name implies?