carla-simulator/carla #620: add a new map

JiaxiongQ, 2018-07-22:
I bought a map from a market, can I add it into carla directly  ?

TheNihilisticRobot, 2018-07-24:
Short answer: yes
The "how" depends on how the map you bought is organised. Please tell me if you need help implementing the map you bought.

KevinYuk, 2018-07-25:
@JiaxiongQ What kind of map do you buy?

JiaxiongQ, 2018-07-25:
@KevinYuk http://www.polypixel3d.com/urban-city-3d-environment-model-pack/ this map

z1223343, 2018-07-27:
have the same question!

z1223343, 2018-07-31:
@TheNihilisticRobot I want to implement this [freeway map](https://www.turbosquid.com/3d-models/3d-highway-streets-construction-sidewalk-model/582579) in carla. I build carla in ubuntu system. There are two questions:
1. How can I implement the freeway map?
2. Is it possible to have NPCs in the new map just like NPCs in town01 and town02 environment?

Thanks! Really appreciate any help you give.

TheNihilisticRobot, 2018-08-01:
You can have pedestrians almost like in our maps but the vehicles are a bit more tricky. Vehicles will be missing the privileged information that our roads give to them (Current position in the lane and direction to follow) so won't automatically follow the road like they do in maps using our roads.

@z1223343 The package you linked in the issue seems rather like a set of pieces than a full map (unlike @JiaxiongQ 's one) . Once you have the models you want to add to your level downloaded you just need to import them to your Carla Unreal project. If you meant to use semantic segmentation (and for best latter organisation), please try to follow Carla's folder system, saving each item in the corresponding folder. Once you have added every piece you need into the project you can create your map and build it.
[UnrealDocumentation](https://docs.unrealengine.com/en-us/Engine/QuickStart)
Once built You'll need to place the corresponding Spawners If you want Npcs in the world. As I've said before, vehicles won´t drive themselves without our roads but you can trick them to follow a Spline using our routeplanners. pedestrians will also need NavMeshes placed in the world to properly work.

Here are some old Issues related to implement and create custom Maps in Carla to further help you:
#636 #504 #160 #603

TheNihilisticRobot, 2018-09-03:
Are you using our spawners to place the vehicles?
If you are using a custom spawner of your own you might be missing giving them a controller.

chengjinjack, 2018-09-20:
@JiaxiongQ 
hello.
Excuse me, how to  add a map into cala?
Thanks

TheNihilisticRobot, 2018-09-20:
It depends on many things. 

Are you trying to create a map? to add one you bought from the unreal store? 
At what point of the map creation do you need assistance?

chengjinjack, 2018-09-21:
@TheNihilisticRobot 
Hi.
I am trying to modify an existing map.The method (https://carla.readthedocs.io/en/stable/map_customization/) introduced on the official website is not very detailed,So I am confused, I don’t know how to proceed.
should I be familiar with the Unreal Engine Editor? Please give me some guidance .
Thanks!

TheNihilisticRobot, 2018-09-21:
Our [Documentation](https://carla.readthedocs.io/en/stable/map_customization/) only Explains how to work with our tools and system. To create a map you'll need to work with the unreal toolset so of course you'll need to learn a bit about the Unreal Editor. There is a lot of good information and tutorials on the web about starting to work with Unreal that will teach you better than I'll ever could how to use the software. 

I'll give you a little headstart:
[UnrealDocumentation](https://docs.unrealengine.com/en-us/)
[LevelCreation quickstart guide](https://docs.unrealengine.com/en-US/Engine/QuickStart) (From Epic's Unreal Documentation)
[Videotutorials on level creation ](https://www.youtube.com/watch?v=cl_eoVfNDKU&list=PLZlv_N0_O1gak1_FoAJVrEGiLIploeF3F)(From Epic's own Youtube channel)

chengjinjack, 2018-09-23:
Hello,@TheNihilisticRobot
Thank you for your patience.
I want to  use  defaultmap provided by the carla to create a simple map ,the map does't have  the PostProcessVol and CarlaMapGenerator.,so how should I add PostProcessVol and CarlaMapGenerator to level ? In addition,when I build the map ,the UE4 always shows that light build failed,how can I solve this problem?

closed 2018-12-01 by stale[bot]