carla-simulator/carla #621: Semantic Segmentation Network

mhusseinsh, 2018-07-23:
Hello,

[image]

You mentioned here on your paper that you trained a semantic segmentation network on dataset from CARLA. Is there by any chance a possibility that you can have this network published, or an open-source code ?

I want also to train a semantic segmentation network on CARLA dataset, and I didn't find a good one to start with that I can adapt on CARLA. I mean I found ones for cityscape but it is difficult to get it adapted on CARLA.

felipecode, 2018-07-23:
It is not yet published. Unfortunately.
But some people have been able to adapt semantic segmentation networks successfully
before like the ErfNet.
Check this paper:
https://arxiv.org/abs/1804.09364

mhusseinsh, 2018-09-21:
@felipecode Thanks a lot
If there will be any update with a Semantic Segmentation network published by your side, then this will be really great

closed 2018-09-21 by mhusseinsh