carla-simulator/carla #629: Example C++ Client Code

p-schulz, 2018-07-26:
For the older versions of carla, without the new network architecture, I wrote a small client. While it's C++ there is some major performance boost compared to python. I thought this could be interesting for some people using the older versions, so I'm sharing it:

https://gist.github.com/p-schulz/a223c4f8fc132424ad084e035aa19bbe

marcgpuig, 2018-07-26:
Hello @p-schulz!

Thank you so much for sharing the code! I can see a lot of work on this :)
Can you check the line 51 of the main.cpp?

```cpp
uint m_world_port = ;
```

I've not tested it, but this is intentioned?

Cheers!

p-schulz, 2018-07-26:
Whoops, indeed... This is not intended, I set it to the default port. The code is an excerpt from my implementation inside a bigger framework and I can't guarantee that it will work out of the box (the functionallity of the main.cpp), but the API works for sure.
Thanks ;-)

hxlee309, 2018-08-11:
Thanks for uploading the cpp client code. Can you tell me how to run the code?

hxlee309, 2018-08-11:
Also, the file "ImageUtils.h" is missing. Could you tell me how to find this file?

p-schulz, 2018-08-13:
@hxlee309 sorry, the file ImageUtils.h isn't used in the code, I removed it from carlaserver.h... 
This code wasn't meant to run out of the box, it's for providing useful methods in order to connect to the server und demonstrating how to use google protobuf to retrieve images, get data from and send controls to the server. You should at least understand what the code is doing as you should paste your own code inside the main routine, e.g. including and using openCV. If you're just using the client to retrieve images or run the simulation, then you would be better off using the python client as it is not as complex as C++.

closed 2018-10-19 by stale[bot]