carla-simulator/carla #634: Expected FPS with LIDAR enabled?

rohanb2018, 2018-07-27:
I have been consistently getting low (below real-time) frame rates when trying to run CARLA with LIDAR enabled, and I want to see whether this is the expected behavior with the CARLA LIDAR sensor or whether there is something with my CARLA/machine setup that is awry.

My software setup involved running a server instance from the Unreal Editor, and connecting to the server using the `carla_ros_bridge` ROS client.

-----------------------

CARLA server configuration:
- Synchronous mode: True
- Quality Level: Low
- Sensors: LIDAR only
- Map: Town01
- Vehicles: 1, Pedestrians: 10

The LIDAR configuration that I used was similar to that of an HDL-64E LIDAR:

```
lidar_0:
  SensorType: LIDAR_RAY_CAST
  carla_settings:
    Channels: 64
    Range: 120
    PointsPerSecond: 1000000
    RotationFrequency: 20
    UpperFovLimit: 2.0
    LowerFovLimit: -24.8
    PositionX: 0.0
    PositionY: 0
    PositionZ: 1.50
    RotationPitch: 0
    RotationYaw: 0
    RotationRoll: 0
```

The frame rates that I was getting, as a function of the CARLA server fixed FPS, are as follows:

**Fixed FPS** | **Measured FPS**
--- | --- 
20 | 4.950
15 | 3.344
10 | 2.463
5 | 1.238

Machine spec:
- GPU: Geforce GTX1080
- CPU: Intel Core i7-7820X
- OS: Ubuntu 16.04

-----------------------
Other observations:

- I tried disabling the rendering, and that didn't have much of an effect. 
- I noticed that the frame rate increased as the number of LIDAR points decreased, indicating that the ray-casting used for the CARLA LIDAR might be the bottleneck. 
- CARLA ran at anywhere from 2-5x real-time when I enabled the camera only.

Any help would be appreciated, and I'd be happy to provide more information about my setup and system specs if that would be helpful. Thanks!

Camouflager, 2018-07-28:
I ran into a similar problem when trying to get more lidar points (fps gets stuck at 2.5 when I try to get more than 500k points per sec). I wonder if there is any way to accelerate this (maybe like parallelizing raycasts). Any suggestion would be appreciated.

nsubiron, 2018-07-28:
@rohanb2018 Thanks for the very detailed report!

With the current Lidar implementation, each point is generated by a ray-cast. Although ray-casts are very optimized in UE4 they are an expensive operation. These sort of implementation is slow with very high number of points per second (with 100k points per second already slows down the simulation about the same as a camera). Some of the things we could try is parallelizing the individual ray-casts or sending them in batches, but the ray-cast API is limited.

However, we plan to add in the future a different implementation of a Lidar. Using the z-buffer fully implemented in the GPU, similar to our depth camera implementation.

Another thing that caught my attention is that there is a significant frame drop when using low fixed FPS. I will take a look into it. Some loss of FPS is expected though, but it seems a bit too much. We have enabled physics substepping, so when the frame rate is very low the physics engine gets updated several times per frame to keep the simulation accurate.

rohanb2018, 2018-07-30:
@nsubiron Thanks for your response! Looking forward to trying out the z-buffer implementation of LIDAR once it is complete.

caiuspb, 2018-08-01:
@nsubiron Do you think it will be possible to simulate a Velodyne Sensor (360° FOV) with a z-buffer implementation? And what about the angular resolution? Would it be configurable or more or less be defined by the 'camera'?
This approach will perform very well in terms of performance but I wonder if it is suitable for lidar simulation.