carla-simulator/carla #635: Empty level can't apply carla plugin

yangzuyuanhao, 2018-07-27:
Hi all, I got a problem when creating a new map. 
When I directly create a new empty level(not duplicate from demo) under the same folder, carla can't plug into it. Anybody knows what setting changes required? Thanks!

nsubiron, 2018-07-28:
Hi @yangzuyuanhao, could you be more specific on what steps you followed and what sort of errors are you getting?

yangzuyuanhao, 2018-07-28:
@nsubiron Hi, I created a new level in the Carla demo project, but when I click play, the ego car and Carla parameters won't show up on editor, and client couldn't receive anything from server.
Attachment 1 is the Carla demo, and 2 is my new level. 
[image]
[image]

nsubiron, 2018-07-28:
Check out the output log to see if any errors appear. Have you put _player starts_ in your map? those are necessary to spawn vehicles.

yangzuyuanhao, 2018-07-28:
@nsubiron  I found the route reason is that my new level missed CarlaMapGenerator. 
So, must I use carlamapgenerator to create road?
But I want to create my own map, not randomly generated by map generator.

yangzuyuanhao, 2018-07-28:
It seems to be another issue? I submitted it in #636

closed 2018-07-28 by yangzuyuanhao