carla-simulator/carla #637: car don't move when I try to driver it in standalone mode

scotthuang1989, 2018-07-29:
I downloaded CARLA_0.7.0.tar.gz.  Run script: CarlaUE4.sh , **the GUI show up, I can control the camera with up and down arrow key, I can see the people are walking, other cars car driving, But I can't move my car with WASD.** 
What could be the problem? I check the output of script, no errors, where should I look to solve this problem?

my OS is ubuntu 17.10, I have 4 Nvidia 1080Ti installed.

marcgpuig, 2018-07-29:
Hi @scotthuang1989.
We experimented similar cases using TeamViewer, is this your case?

scotthuang1989, 2018-07-29:
yes. I am using teamviewer, I just try it on local computer, it works.. but My local computer install ubuntu 16.04 ( teamview os is ubuntu 17.10)

marcgpuig, 2018-07-29:
Ok so I'm closing the issue because we localized the problem here.

closed 2018-07-29 by marcgpuig