carla-simulator/carla #638: Problem when building on Win / Compiling the .dll s

grafandreas, 2018-07-30:
Hi all,

I am following the instructions for building on Win (Win 10, 64bit). Everything seems to run through until Unreal tells me about missing .dlls and unsuccessfully tries to rebuild. The log excerpt is:

Running C:/Program Files/Epic Games/UE_4.18/Engine/Binaries/DotNET/UnrealBuildTool.exe CarlaUE4 Development Win64 -project="C:/git/carla2/carla/Unreal/CarlaUE4/CarlaUE4.uproject" -editorrecompile -progress -NoHotReloadFromIDE
Creating makefile for CarlaUE4 (no existing makefile)
Performing full C++ include scan (no include cache file)
ERROR: Unable to instantiate module 'Carla': System.NotImplementedException: Die Methode oder der Vorgang ist nicht implementiert.
          bei Carla.AddCarlaServerDependency(ReadOnlyTargetRules Target) in c:\git\carla2\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Carla.Build.cs:Zeile 98.
       (referenced via CarlaUE4.uproject -> Carla.uplugin)

Any idea what could be wrong?

Thank you very much,

Andreas

iFuSiiOnzZ, 2018-07-30:
Currently the Windows build is not available in master. We are still testing and fixing problems during the build, but if you wish to test it there is a branch called 0.9-win32 where you can find.

`` `
git clone https://github.com/carla-simulator/carla.git
git checkout 0.9-win32
make launch
`` `

You will have to manually download the assets from https: //drive.google.com/open? Id = 1FtC00CrDb7Kz5StBAwb6vqOGbzZtpROx` and unzip them into `Unreal / CarlaUE4 / Content` (create it if it does not exist), it would be a good idea doing it before you call `make launch` will also start the Unreal Editor.

grafandreas, 2018-08-02:
Thanks for the information and the help.

with the instructions above, I am now seeing the following in the build log of Unreal:

Module.Carla.cpp
C:\git\carla3\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp(252): error C2440: 'initializing': cannot convert from 'initializer list' to 'FActorAttribute'
C:\git\carla3\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp(252): note: No constructor could take the source type, or constructor overload resolution was ambiguous
C:\git\carla3\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp(252): error C2672: 'TArray<FActorAttribute,FDefaultAllocator>::Emplace': no matching overloaded function found
C:\git\carla3\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\rpc/msgpack/predef/other/endian.h(151): warning C4668: 'MSGPACK_ARCH_AMD64' is not defined as

and

C:\git\carla3\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Walker\WalkerAIController.cpp(161): error C2664: 'void TBaseDynamicMulticastDelegate<FWeakObjectPtr,void,TArray<AActor *,FDefaultAllocator>>::__Internal_AddDynamic<AWalkerAIController>(UserClass *,void (__cdecl AWalkerAIController::* )(TArray<AActor *,FDefaultAllocator>),FName)': cannot convert argument 2 from 'void (__cdecl AWalkerAIController::* )(const TArray<AActor *,FDefaultAllocator> &)' to 'void (__cdecl AWalkerAIController::* )(TArray<AActor *,FDefaultAllocator>)'
        with
        [
            UserClass=AWalkerAIController
        ]
C:\git\carla3\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Walker\WalkerAIController.cpp(161): note: Types pointed to are unrelated; conversion requires reinterpret_cast, C-style cast or function-style cast

(I am running the "make launch" from a x64 native tools vs 2017" command shell on Win10.

iFuSiiOnzZ, 2018-08-02:
Hi @grafandreas 

Strange, have you tried to build it from the VS2017? Because I am wondering if you have other Visual Studio installed and Unreal builds it with that other one instead of 2017.

For generating the Visual Studio solution, right click over `Unreal\CarlaUE4\CarlaUE4.uproject -> Generate Visual Studio project files` and after open with VS2017 the `Carla.sln`

Make sure you have:
- Unreal Engine 4.19
- Windows SDK 10

Also create a new Unreal project and open it to see if everything works fins, as Unreal will give errors if something is mising or wrong configured. To go faster with the issue, can you please connect to carla discord?

PS: We move to #647 to have it everything better organized

Greetings

closed 2018-08-20 by iFuSiiOnzZ