carla-simulator/carla #643: make: ** no rule to make target 'clean'. Stop

BrianN92, 2018-07-31:
Hi all, I am new to Carla. I met an error when running "Rebuild.bat". It shows "make: ** no rule to make target 'clean'. Stop. 'Not implemented!' ". How should I solve it? Thanks!

vcshih, 2018-08-01:
Ran into the same issue and I'm assuming you're trying to build on Windows. According to the 0.9.0 docs:
```Since version 0.9.0 the build system changed, the instructions in this document are no longer valid. While we work on fixing this, we recommend either compile Carla on Linux or switch to the stable branch. Sorry for the inconvenience.```
And if you look into the Windows.mk in carla/Util/BuildTools/, it seems like the Windows makefile has not been implemented yet, so your best bet is to use one of the older versions for now.

iFuSiiOnzZ, 2018-08-02:
@BrianN92 Any other problems post it to #647 please

closed 2018-08-20 by iFuSiiOnzZ