carla-simulator/carla #644: Should pixel density be 0.1653 or 0.1643?

rowanmcallister, 2018-07-31:
Should pixel density be `0.1653` or `0.1643`?

0.1653 here:
https://github.com/carla-simulator/carla/blob/be31711d5146cbd8785b55096d56a550a35879e6/PythonClient/view_start_positions.py#L42

0.1643 here:
https://github.com/carla-simulator/carla/blob/be31711d5146cbd8785b55096d56a550a35879e6/PythonClient/carla/planner/city_track.py#L19

I first though this value was arbitrary. But because pixel_density seems important (i.e. defining how a world point converts to a node), I'm worried my world-defined waypoints are slightly misplaced?

danil-tolkachev, 2018-08-01:
I've also noticed this when I make PR #542. `0.1643` seems a correct value.

nsubiron, 2018-08-01:
PR merged!

closed 2018-10-07 by stale[bot]

zjnyly, 2023-04-05:
what is the distance measurement for pixel density, in `cm` or in `mm`?