carla-simulator/carla #646: Some Questions About Autopilot

yuanzhiyu, 2018-08-02:
Hey! 
Recently I'm trying a pipeline-way to control a car in the simulators. So I have read a part of the source codes of autopilot. But I still have some questions.
1. Which method is used to control a car to turn at an intersection? Does "CalcStreeringValue(Direction)" solve this problem? 
2. What does TargetLocation mean? I understand it as a planned route consist of a series points vector. But why it executes "CalcStreeringValue(Direction)" when "TargetLocation is empty". if I can understand the TargetLocation as a straight-line route without any turning?
3. How can I make sure that the car has reached a point on the planned route and to compute the "steering" with the next point? It should have a method to judge whether the car has reached a point, but I haven't found it. 
4. How does autopilot create a planned route automatically and update the variable “Location”?

Thank you !!

yuanzhiyu, 2018-08-02:
Also, a question does not belong to autopilot, where can I find the relationship between start point index and its coordinate in the map? Should it be in a file?

shriram17, 2018-08-03:
Hey @yuanzhiyu 
From my understanding, when Autopilot Mode is switched on,

- The car proceeds to move in a forward direction at the speed which is set by the speed limit for the given region (default - 30 km/hr)
- The car stops when it detects the presence of other cars or pedestrians in its surroundings and if it encounters a red traffic light achieved using ray tracing and trigger overlap information (see WheeledVehicleAIController)
- The car uses the roadmap data to steer itself to ensure it stays on the right side of the road and also ensures that car doesn’t enter the sidewalk/footpath. (see CalcStreeringValue(Direction) in WheeledVehicleAIController)
- When the car enters an intersection, an overlap trigger event calls the corresponding Route Planner (visible in Town maps) which provides the car a route as a set of spline points that determine the direction (left/right/straight) the car will head towards. These spline points are sent to the VehicleController through the SetFixedRoute() as the 'TargetLocations'. (See [RoutePlanner](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/RoutePlanner.cpp)) 
The routes are picked randomly using the RandomEngine based on the probabilities that have been set for the routes. (These probabilities can be changed in the editor or in-game.)

I hope this answers some or most of your questions.

yuanzhiyu, 2018-08-07:
@shriram17 
Thanks a lot for answering my question! But still, the algorithm on autopilot is not easy to understand, I choose pure-pursuit algorithm to solve my problem finally.

Neel1302, 2020-01-19:
@shriram17 
How can the probabilities for the splines be set in-game?
I know it can set in OpenDriveActor in the Map before launching the simulator, but in scenarios when one actor needs to turn right and the other left, it doesn't work out. So it is best to set it in-game - any pointers as to how this can be done would be greatly appreciated.
Thanks