carla-simulator/carla #647: Building on Windows 0.9.X

iFuSiiOnzZ, 2018-08-02:
# New and automated instructions
Make sure you have:
```
Unreal Engine 4.19
Visual Studio 2017 (as default version, if not set it) 
Windows SDK 10 (should be installed with VS2017)
```

Instructions:
```
1) git clone https://github.com/carla-simulator/carla.git
2) git checkout 0.9-win32

3) Download https://drive.google.com/uc?id=1FtC00CrDb7Kz5StBAwb6vqOGbzZtpROx&export=download
4) Extract in Unreal/CarlaUE4/Content/Carla

5) make launch
```

## Before posting in this issue:
- Search for **similar comments** (just a quick `ctrl+F` search).
- Make sure you use the **correct Unreal Engine version**.

If your questions satisfy the previous points, we will be glad to answer them. :grimacing:

## Discord
Consider join our discord channel for a further talk.
We have a Windows specific channel.
[image]

- - -

## Manual build
Follow the instructions from 1 to 4, at this point, double-clicking on `Unreal/CarlaUE4/CarlaUE4.uproject` should work. A warning about Carla plugin being not compatible appears sometimes, it’s an Unreal’s bug and can be ignored, press `no` to disable Carla plugin. Then press `yes` when asked to compile the missing modules.

If that doesn’t work, you can open the project with Visual Studio. Right click the `CarlaUE4.uproject` and select `Generate Visual Studio Project Files`. Then open the Visual Studio solution it generates (*.sln). Hopefully this will give some hints on what fails.

- - -

PS: If you are coming from an open issue related with the build please make reference to it

grafandreas, 2018-08-02:
Is it Windows 10 SDK? I just tried on a PC after new installation of VC 2017 and Unreal 4.19 and the Unreal log says:

Reflection code generated for CarlaUE4Editor in 13,6607113 seconds
@progress pop
ERROR: Windows SDK v8.1 must be installed in order to build this target.

iFuSiiOnzZ, 2018-08-03:
Hi @grafandreas 

I took a look at the Visual Studio setup for the Unreal Engine, and yes, it seems they ask for SDK 8.1.
You can install it using the `Visual Studio Installer`, press `Modify`, under the tab `Individual componentes` search for `SDKs, libraries and framework` and check the `Windows 8.1 SDK`, for last press `Modify`.

BrianN92, 2018-08-03:
I also failed to compile it. The errors are listed below: 

Severity	Code	Description	Project	File	Line	Suppression State
Error	C2440	**'initializing': cannot convert from 'initializer list' to 'FActorAttribute'**	CarlaUE4	D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp	252	

Severity	Code	Description	Project	File	Line	Suppression State
Error	C2672	**'TArray<FActorAttribute,FDefaultAllocator>::Emplace': no matching overloaded function found**	CarlaUE4	D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp	252

Severity	Code	Description	Project	File	Line	Suppression State
Error		**UBT ERROR: Failed to produce item:** D:\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-Carla-Win64-DebugGame.lib	CarlaUE4	D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\UnrealBuildTool	1	

Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3075	The command "D:\Epic\UE_4.19\Engine\Build\BatchFiles\Build.bat CarlaUE4Editor Win64 DebugGame "D:\carla\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.	CarlaUE4	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets	41

grafandreas, 2018-08-04:
@BrianN92 I experienced this too and it went away after installing VC 2017 and everything on a PC without a previous VC installation. Note that, if you have more than one VC installation on your PC, maybe an older version is being used during build (IIRC, the build files ask for VC 2015 as minimum, but that is not sufficient).

iFuSiiOnzZ, 2018-08-20:
If during the compilation you are getting an error like `initializing': cannot convert from 'initializer list' to 'FActorAttribute` please update Visual Studio to the latest version as the compiler of your current version of VS2017 doesn't support that kind of initialization.

iFuSiiOnzZ, 2018-08-20:
@overlapjho Have you manage to fix it? It seems like is missing all the dependencies. Sorry for the delay in response, we have been in vacations.

overlapjho, 2018-08-20:
@iFuSiiOnzZ not yet, I've tried several solutions from other issues did not work

iFuSiiOnzZ, 2018-08-20:
@overlapjho take a look if under `Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies` you have the next folder structure:

```
includes
    - boost
    - carla
    - compiler
    - rpc

lib
    - carla_server.lib
    - carla_server_debug.lib
    - libboost_date_time-vc141-mt-x64-1_68.lib
    - libboost_system-vc141-mt-x64-1_68.lib
    - rpc.lib
```

overlapjho, 2018-08-20:
@iFuSiiOnzZ  yes I have those, this is my recent error product when i try to build it earlier 

> 1>------ Rebuild All started: Project: UE4, Configuration: BuiltWithUnrealBuildTool Win32 ------
> 2>------ Rebuild All started: Project: CarlaUE4, Configuration: Development_Editor x64 ------
> 2>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.props(31,3): warning MSB4011: "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Makefile.props" cannot be imported again. It was already imported at "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.cpp.props (31,3)". This is most likely a build authoring error. This subsequent import will be ignored. [C:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
> 1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.props(31,3): warning MSB4011: "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Makefile.props" cannot be imported again. It was already imported at "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.cpp.props (31,3)". This is most likely a build authoring error. This subsequent import will be ignored. [C:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
> 2>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.props(31,3): warning MSB4011: "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Makefile.props" cannot be imported again. It was already imported at "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.cpp.props (31,3)". This is most likely a build authoring error. This subsequent import will be ignored. [C:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\UE4.vcxproj]
> 1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Cpp.props(31,3): warning MSB4011: "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.Makefile.props" cannot be imported again. It was already imported at "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.cpp.props (31,3)". This is most likely a build authoring error. This subsequent import will be ignored. [C:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\UE4.vcxproj]
> 2>Cleaning CarlaUE4Editor Binaries...
> 2>Creating makefile for CarlaUE4Editor (no existing makefile)
> 2>Performing full C++ include scan (no include cache file)
> 2>Parsing headers for CarlaUE4Editor
> 2>  Running UnrealHeaderTool "C:\carla\Unreal\CarlaUE4\CarlaUE4.uproject" "C:\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\CarlaUE4Editor\Development\CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
> 2>Reflection code generated for CarlaUE4Editor in 53.2410485 seconds
> 2>Performing 17 actions (2 in parallel)
> 2>PCH.CarlaUE4.cpp
> 2>SharedPCH.Engine.cpp
> 2>[3/17] Resource ModuleVersionResource.rc.inl
> 2>Module.Carla.cpp
> 2>C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp(252): error C2440: 'initializing': cannot convert from 'initializer list' to 'FActorAttribute'
> 2>C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp(252): note: No constructor could take the source type, or constructor overload resolution was ambiguous
> 2>C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp(252): error C2672: 'TArray<FActorAttribute,FDefaultAllocator>::Emplace': no matching overloaded function found
> 2>C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\rpc/msgpack/predef/other/endian.h(151): warning C4668: 'MSGPACK_ARCH_AMD64' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
> 2>C:\Program Files (x86)\Windows Kits\8.1\include\um\winnt.h(17005): warning C4668: '_APISET_RTLSUPPORT_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
> 2>C:\Program Files (x86)\Windows Kits\8.1\include\um\interlockedapi.h(73): warning C4668: '_APISET_INTERLOCKED_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
> 2>C:\Program Files (x86)\Windows Kits\8.1\include\um\processthreadsapi.h(1170): warning C4668: '_WIN32_WINNT_WINTHRESHOLD' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
> 2>C:\Program Files (x86)\Windows Kits\8.1\include\um\securitybaseapi.h(357): warning C4668: '_APISET_SECURITYBASE_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
> 2>C:\Program Files (x86)\Windows Kits\8.1\include\um\securitybaseapi.h(472): warning C4668: '_APISET_SECURITYBASE_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
> 2>C:\Program Files (x86)\Windows Kits\8.1\include\um\securitybaseapi.h(1302): warning C4668: '_APISET_SECURITYBASE_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
> 2>C:\Program Files (x86)\Windows Kits\8.1\include\um\winbase.h(8618): warning C4668: 'NTDDI_WIN7SP1' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
[... cut ...]
> 2>UnrealBuildTool : error : UBT ERROR: Failed to produce item: C:\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-Carla.lib
> 2>Total build time: 567.10 seconds (Local executor: 0.00 seconds)
> 2>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(46,5): error MSB3073: The command "D:\UE_4.19\Engine\Build\BatchFiles\Rebuild.bat CarlaUE4Editor Win64 Development "C:\carla\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild" exited with code -1.
> 2>Done building project "CarlaUE4.vcxproj" -- FAILED.
> ========== Rebuild All: 1 succeeded, 1 failed, 0 skipped ==========

iFuSiiOnzZ, 2018-08-20:
@overlapjho As compilation errors I am seeing `error C2440` and `error C2672`, all the others are warnings. Make sure  you have the latest version of VS2017 (as the current version of VS2017 that you have the compiler doesn't support those features).

Can you also specify wich version do you have? On Visual Studio go to `Help` -> `About Microsoft Visual Studio`, mine is 15.7.6

iFuSiiOnzZ, 2018-08-20:
@overlapjho Do you have any other versión of Visual Studio intalled? If you have it can you uninstall it? It should give a few warning but any error, strange.

PS: Can you join discord to make the conversation more fluent?

iFuSiiOnzZ, 2018-08-24:
If you also have installed other version of Visual Studio or Microsoft Compiler and get any error during the build as `C2440` or `C2672` this is probably because is not using the 2017 compiler, so try first to uninstall the other ones and rebuild again.

If you would like to keep the other version of Visual Studio edit `%appdata%\Unreal Engine\UnrealBuildTool\BuildConfiguration.xml` adding the following lines:

```
<VCProjectFileGenerator>
    <Version>VisualStudio2017</Version>
</VCProjectFileGenerator>

<WindowsPlatform>
    <Compiler>VisualStudio2017</Compiler>
</WindowsPlatform>
```

M-Hammod, 2018-09-05:
Hi, 

I've successfully built carla using the instruction above (windows) , also I was able to get hit play or run CarlaUE4.exe stand alone. but when I ran carla\PythonClient\manual_control unreal crashes and I get the following error 

[image]

[image]

iFuSiiOnzZ, 2018-09-06:
Hi @M-Hammod 

We tried to reproduce the error but we could not, can you try to compile and execute Carla from Visual Studio and when it crashes try to provide more information about the crash (call stack of Carla, etc). It seems like the some exception generated from the rpclib it wasn't captured.

M-Hammod, 2018-09-06:
Hi @iFuSiiOnzZ, 
it seems that there is a difference between the manual_control.py under PythonAPI and PythonClient, this problem went away when I used PythonAPI code, also I have to run setup.py build & install. 
Thanks.

iFuSiiOnzZ, 2018-09-06:
Thank you for the feedback @M-Hammod, `PythonAPI is for Carla >= 0.9` and `PythonClient for Carla < 0.9`. In the future when the 0.9.X will be stable, legacy files it will be removed.

gut-yu, 2018-09-12:
@overlapjho @grafandreas @iFuSiiOnzZ 
Hi
I have met the same problem.
“E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\CarlaUE4.sln”(Default target) (1) -> “E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj”(Default target) (2) ->
 (Build target) ->
 E:\CARLA\carlamaster\carlamaster\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp(252): error C2440: “initializing”: cannot convert from 'initializer list' to 'FActorAttribute' [E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
 E:\CARLA\carlamaster\carlamaster\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorBlueprintFunctionLibrary.cpp(252): error C2672: 'TArray<FActorAttribute,FDefaultAllocator>::Emplace': no matching overloaded function found
 [E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
UnrealBuildTool : error : UBT ERROR: Failed to produce item: E:\CARLA\carlamaster\carlamaster\Unreal\CarlaUE4\Plugins\Carla\Binaries\Win64\UE4-Carla.lib [E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3075: The command“"D:\AirSim Project\ue4_4.19.2\UE_4.19\Engine\Build\BatchFiles\Build.bat" CarlaUE4 Win64 Development "E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild”exited with code 5. Please verify that you have sufficient rights to run this command.
 [E :\CARLA\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]

I ever installed Visual Studio 2015. I have tried many ways to remove it compeletely. 
But it does not work.
Help!

iFuSiiOnzZ, 2018-09-12:
Hi @gut-yu 

Have you tried to edit `%appdata%\Unreal Engine\UnrealBuildTool\BuildConfiguration.xml` setting the compiler and the version to `VisualStudio2017`? On the first post you have what you have to add to the file. Make a clean before before recompile again.
 ```
make clean
make launch
```

If is not working edit the `Util\BuildTools\BuildCarlaUE4.bat` and on line 134 add `-2017` after `-game -rocket -progress`. the line should be like this `-game -rocket -progress -2017` now

gut-yu, 2018-09-12:
@iFuSiiOnzZ 
I've added the lines.It works.
However,there is still an error as followed :
`“E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\CarlaUE4.sln”(Default target) (1) ->
  “E:\CARLA\carla-master\carlamaster\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj”(Default target) (2) ->
       (Build target) ->
         UnrealBuildTool : error : multiple root elements. Line 4, location 2. [E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3073: command“"D: \AirSim Project\ue4_4.19.2\UE_4.19\Engine\Build\BatchFiles\Build.bat" CarlaUE4 Win64 Development "E:\CARLA\carla
       -master\carla-master\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild”exited with code 1.  [E:\CARLA\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]`
What should I do?

iFuSiiOnzZ, 2018-09-12:
First time that I see that error, you XML file look like this?

```
<?xml version="1.0" encoding="utf-8" ?>
<Configuration xmlns="https://www.unrealengine.com/BuildConfiguration">
    <VCProjectFileGenerator>
        <Version>VisualStudio2017</Version>
    </VCProjectFileGenerator>

    <WindowsPlatform>
        <Compiler>VisualStudio2017</Compiler>
    </WindowsPlatform>
</Configuration>
```

gut-yu, 2018-09-12:
I just added the lines like this:
`<?xml version="1.0" encoding="utf-8" ?>
<Configuration xmlns="https://www.unrealengine.com/BuildConfiguration">
</Configuration>
<VCProjectFileGenerator>
    <Version>VisualStudio2017</Version>
</VCProjectFileGenerator>

<WindowsPlatform>
    <Compiler>VisualStudio2017</Compiler>
</WindowsPlatform>`
Anything wrong?

gut-yu, 2018-09-12:
I just added these lines at the end :
[image]
Anything wrong?

gut-yu, 2018-09-12:
I changed the position of these lines as you showed me .
It works !!!
Thanks a lot ! @iFuSiiOnzZ

iFuSiiOnzZ, 2018-09-12:
You are welcome @gut-yu : )

mshaheerm, 2018-09-28:
i tried 'make clean' and then 'make launch-editor' but the build fails with this error. how to fix it?
[image]

iFuSiiOnzZ, 2018-10-01:
Hi,

Hope you found a solution, if not, try to manually delete delete the `boost-install` folder from `Build` (and if you see that the problem still persist also delete `boost-src`) and try again.

adarsh9975, 2018-10-01:
Ok so i solved the above problem by moving my contents to a folder with folder-name having no spaces.
But now it says it cant find "CMakeLists.txt"

adarsh9975, 2018-10-01:
I copied CMakeLists.txt from parent folder to the  "rpclib_src" folder and it started compiling however it stopped with following message:

D:\Self_driving_Car_setup\CARLA_Project\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  4
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "D:/Self_driving_Car_setup/CARLA_Project/carla/Build/"
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "D:/Self_driving_Car_setup/CARLA_Project/carla/Build/"
    -[install_rpclib]: Not cloning rpclib because already exists a folder called "rpclib-src".
    -[install_rpclib]: Generating build...
-- The C compiler identification is MSVC 19.15.26730.0
-- The CXX compiler identification is MSVC 19.15.26730.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:4 (include):
  include could not find load file:

    Build/CMakeLists.txt.in

CMake Error at CMakeLists.txt:6 (add_subdirectory):
  add_subdirectory given source "LibCarla/cmake" which is not an existing
  directory.

-- Configuring incomplete, errors occurred!

iFuSiiOnzZ, 2018-10-01:
Hi @adarsh9975 

Delete the Build folder and try again. That file should be created automatically by setup.bat (Util\BuildTools).

adarsh9975, 2018-10-01:
Thank you @iFuSiiOnzZ I just completed my build process. The build completed successfully how ever when i launch the CarlaUE4.exe i get this:

[image]

obviously build was not proper. Build did complete successfully but i had a warning that some parts of unreal engine had to be re-built (which completed automatically successfully),also there were a lot of warnings during the unreal-rebuild. What should i do now ? I am using Unreal engine 4.19.2

adarsh9975, 2018-10-01:
I opened the project in unreal editor. Looks like the assets can't be located. However I have downloaded the content and extracted it to " Unreal\CarlaUE4\Content\Carla ". I am using 0.8.4 content for CARLA.

This is the content folder:
[image]

Any help will be greatly appreciated.

iFuSiiOnzZ, 2018-10-01:
You need the content for the 0.9.X as master branch is on 0.9.X. In case you need the 0.8.X you will have to checkout it.

adarsh9975, 2018-10-02:
Seems like the assets in content folder are arranged differently for 0.8.x and 0.9.x (for example compare Blueprints/Game for both versions). I downloaded the content for 0.9.x and built the code again. Now it runs the environment. Thanks for your help @iFuSiiOnzZ

mshaheerm, 2018-10-03:
[image]

what are these errors for? and how to fix them?

iFuSiiOnzZ, 2018-10-03:
@mshaheerm that is because you didn't download Carla contents, take a look at [Download Carla contents](https://carla.readthedocs.io/en/latest/how_to_build_on_windows/).

mshaheerm, 2018-10-04:
@iFuSiiOnzZ  thanks alot. fixed it.

mshaheerm, 2018-10-06:
@iFuSiiOnzZ  the problem with the boost-install still exists. in the Build folder there isnt any boost-install folder. only boost-src which i removed and tried again but the problem was still there.

iFuSiiOnzZ, 2018-10-08:
@mshaheerm can you provide more information of what happened during the build, any log or on screen information.

SumerHoT, 2018-10-25:
Branch 0.9-win32 is missing.
In Sep, I successed builing Carla following above steps, while recently I failed to repeat the steps, since I could not checktou the 0.9-win32 branch. What should I do now?

SumerHoT, 2018-10-25:
@iFuSiiOnzZ Thanks! So I could not blame on this and keep fighting with the linking issues now... 0_0

SumerHoT, 2018-10-25:
Hi @iFuSiiOnzZ,  While make launch, the command line report:
D:\Carla\carla\LibCarla\cmake\..\source\carla/sensor/CompileTimeTypeMap.h(97): error C2975: “Size”:“carla::sensor::detail::CompileTimeTypeMapImpl”的模板参数无效，应为编译时常量表达式 (编译源文件 D:\Carla\carla\LibCarla\source\carla\sensor\Deserializer.cpp) [D:\Carla\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\carla_server.vcxproj]
The Chinese means "the template parameters are invalid, should be constexpr(compling the source file)"

That happens in 2 files and result in total 4 errors:
D:\Carla\carla\LibCarla\cmake\..\source\carla/sensor/CompileTimeTypeMap.h(97)
D:\Carla\carla\LibCarla\cmake\..\source\carla/sensor/CompileTimeTypeMap.h(100)

Could you help to figure out what should I do with them?

iFuSiiOnzZ, 2018-10-25:
@SumerHoT
Compilation problems between `clang` and `visual cl`. Change `size()` with `sizeof...(Items)`, we will try to fix it in the future.

SumerHoT, 2018-10-25:
@iFuSiiOnzZ Thanks a lot! That works! 
But while compling CarlaUE4 in Unreal Engine, it shows CarlaUE4 could not be complied, please try manual complile from source.
The Unreal Engine is 4.19.2, and previous machine works fine.
Any idea about this?

iFuSiiOnzZ, 2018-10-25:
Of course is a compilation problem, open the file and change what i told you in the previous post.

SumerHoT, 2018-10-29:
@iFuSiiOnzZ While compling CarleUE4 Unreal Project, two main errors occur:
==============
1: LogCompile : error : Circular dependency detected for filename D:\Carla\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Settings\QualityLevel.h!
2: D:\Carla\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\carla/geom/BoundingBox.h(27): error C2512: “carla::geom::Location” No suitable default constructor available.
==============
In file: D:\Carla\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\carla\geom\Location.h
I add:	  Location() {};
But I couldn't find out the way to solve Error 1#. Any suggestion?

iFuSiiOnzZ, 2018-10-29:
Hi @SumerHoT 

Those 2 errors are quite easy to fix. For the circular dependency you can leave it as it is, it builds any way or change the name of the file to `QualityLevelUE.h` for instances and all the includes where it appears.

For the C2512 error you will have to modify the `Vector3D.h` and `Location.h`, in `Vector3D` you will have to add the copy constructor and for `Location` add the constructors:
```cpp
    Vector3D(const Vector3D &v3d)
      : x(v3d.x),
        y(v3d.y),
        z(v3d.z) {}
```

```cpp
    Location() = default;

    Location(const Location &location)
      : Vector3D(location) {}

    Location(float ix, float iy, float iz)
      : Vector3D(ix, iy, iz) {}
```

Once that you have done that, it will compile the Editor but the game, as it seems that some Unreal functions are overwritten. We are currently working to fix it.

SumerHoT, 2018-10-29:
@iFuSiiOnzZ 
Thanks, I modified above and then build unreal project manually and works.
Do I still need to make PythonAPI now?
Since there still are errors while make PythonAPI:
1#. Traceback (most recent call last):
  File "setup.py", line 101, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 51, in get_libcarla_extensions
    sources += [x for x in walk('dependencies/include/carla', '*.cpp')]
UnboundLocalError: local variable 'walk' referenced before assignment
I move the def walk to above, and then new errors occur:

2#. source/libcarla/libcarla.cpp(7): fatal error C1083: Could not open include file: “carla/Memory.h”: No such file or directory

iFuSiiOnzZ, 2018-10-29:
That is a bit more tricky, for now no one started to port the PythonAPI to Windows. It has some dependencies for png, jpeg and tiff support. So if you want to test the 0.9.X please use linux for now as we are currently busy with the 0.9.1 release. Sorry!

SumerHoT, 2018-10-30:
@iFuSiiOnzZ So could you reopen the 0.9-win32 branch for these days?

iFuSiiOnzZ, 2018-10-30:
I am sorry, but unfortunately the branch is deleted. You could try to checkout some old commit and see if it builds, sadly I can not tell you with which one to start as I don't know in what point the Windows build broken.

SumerHoT, 2018-10-30:
@iFuSiiOnzZ Nevermind and thanks any way, I could try rebuilding from my archived code :) Hope you guys all good~

MaxTD, 2018-11-17:
3>UnrealBuildTool : error : Failed to start local process for action: C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX64\x64\cl.exe  @"C:\Users\carlo\Desktop\carla-master\carla-maste
       r\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4\Development\Carla\Module.Carla.gen.3_of_3.cpp.obj.response" [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
     3>UnrealBuildTool : error : Failed to start local process for action: C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX64\x64\cl.exe  @"C:\Users\carlo\Desktop\carla-master\carla-maste
       r\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4\Development\Carla\Module.Carla.gen.1_of_3.cpp.obj.response" [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
     3>UnrealBuildTool : error : Failed to start local process for action: C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX64\x64\cl.exe  @"C:\Users\carlo\Desktop\carla-master\carla-maste
       r\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4\Development\Carla\Module.Carla.cpp.obj.response" [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
     3>UnrealBuildTool : error : Failed to start local process for action: C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX64\x64\cl.exe  @"C:\Users\carlo\Desktop\carla-master\carla-maste
       r\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4\Development\Carla\Module.Carla.gen.2_of_3.cpp.obj.response" [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
     3>UnrealBuildTool : error : System.ComponentModel.Win32Exception (0x80004005): The system cannot find the file specified [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
                                    at System.Diagnostics.Process.StartWithCreateProcess(ProcessStartInfo startInfo)
                                    at UnrealBuildTool.ActionThread.ThreadFunc()
     3>UnrealBuildTool : error : System.ComponentModel.Win32Exception (0x80004005): The system cannot find the file specified [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
                                    at System.Diagnostics.Process.StartWithCreateProcess(ProcessStartInfo startInfo)
                                    at UnrealBuildTool.ActionThread.ThreadFunc()
     3>UnrealBuildTool : error : System.ComponentModel.Win32Exception (0x80004005): The system cannot find the file specified [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
                                    at System.Diagnostics.Process.StartWithCreateProcess(ProcessStartInfo startInfo)
                                    at UnrealBuildTool.ActionThread.ThreadFunc()
     3>UnrealBuildTool : error : System.ComponentModel.Win32Exception (0x80004005): The system cannot find the file specified [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
                                    at System.Diagnostics.Process.StartWithCreateProcess(ProcessStartInfo startInfo)
                                    at UnrealBuildTool.ActionThread.ThreadFunc()
     3>UnrealBuildTool : error : UBT ERROR: Failed to produce item: C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Plugins\Carla\Binaries\Win64\UE4-Carla.lib [C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\Car
       laUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         Total build time: 6.74 seconds (Local executor: 0.00 seconds)
     3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.19\Engine\Build\BatchFiles\Build.bat" CarlaUE4
        Win64 Development "C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command. [C:\Users\car
       lo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
     3>Done Building Project "C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj" (default targets) -- FAILED.
     1>Done Building Project "C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\CarlaUE4.sln" (default targets) -- FAILED.

Build FAILED.

       "C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\CarlaUE4.sln" (default target) (1) ->
       "C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj" (default target) (3) ->
       (Build target) ->
         UnrealBuildTool : error : Failed to start local process for action: C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX64\x64\cl.exe  @"C:\Users\carlo\Desktop\carla-master\carla-mas
[... cut ...]
         C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.19\Engine\Build\BatchFiles\Build.bat" CarlaU
       E4 Win64 Development "C:\Users\carlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command. [C:\Users\c
       arlo\Desktop\carla-master\carla-master\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]

nicksunyang, 2018-11-17:
“D:\carla\Unreal\CarlaUE4\CarlaUE4.sln”(默认目标) (1) ->
       “D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj”(默认目标) (2) ->
       (Build 目标) ->
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(59): error C2039: “_
       InterlockedCompareExchange”: 不是“FWindowsPlatformAtomics”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\
       CarlaUE4.vcxproj]
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(59): error C2665: “_
       InterlockedCompareExchange”: 4 个重载中没有一个可以转换所有参数类型 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.v
       cxproj]
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(78): error C2039: “_
       _faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxpr
       oj]
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHICommandList.h(282): error C2039:
        “_InterlockedIncrement”: 不是“FWindowsPlatformAtomics”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHICommandList.h(282): error C2665:
        “_InterlockedIncrement”: 4 个重载中没有一个可以转换所有参数类型 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxp
       roj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1689): error C20
       39: “GetObjectW”: 不是“TScriptInterface<IBlendableInterface>”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFil
       es\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1709): error C20
       39: “GetObjectW”: 不是“TScriptInterface<IBlendableInterface>”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFil
       es\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(3
       7): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(4
       5): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       28): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       37): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1689): error C20
       39: “GetObjectW”: 不是“TScriptInterface<IBlendableInterface>”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFil
       es\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1709): error C20
       39: “GetObjectW”: 不是“TScriptInterface<IBlendableInterface>”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFil
       es\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(3
       7): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(4
       5): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       28): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       37): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(103): erro
       r C2039: “__faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(154): erro
       r C2039: “__faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(159): erro
       r C2039: “__faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(165): erro
[... cut ...]
       in64\UE4-Carla.lib [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.T
       argets(44,5): error MSB3075: 命令“"D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Build\BatchFiles\Build.bat" Ca
       rlaUE4 Win64 Development "D:\carla\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild”已退出，代码为 5。 请验证您是否拥有
       运行此命令的足够权限。 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]

iFuSiiOnzZ, 2018-11-19:
Hi all, there is a branch called fix_win32_comilation, the windows version should build now.
@nicksunyang @MaxTD

dieteekanne2, 2018-11-22:
Hi,
i get the following error when I try to run any of the python scritpts (with Carla v0.9.1):
C:\Carla\carla_0.9\carla_fix_win32_Try_2\carla-fix_win32_compilation>python .\PythonAPI\tutorial.py
Traceback (most recent call last):
  File ".\PythonAPI\tutorial.py", line 21, in <module>
    import carla
ImportError: No module named 'carla'

I rebuild the PythinAPI and LibCarla hoping it would help, but it did't. 
Thanks for your help!

iFuSiiOnzZ, 2018-11-23:
For building PythonAPI take a look at the issue #976, once you have built it go to PythonAPI folder and run any of the scripts.

marcgpuig, 2018-11-26:
Hi @dieteekanne2.
Did you compiled the python API?
```
make PythonAPI
```

Atlunch, 2018-11-27:
@iFuSiiOnzZ 
I'm still getting the same errors like @nicksunyang @MaxTD while following the "How to build on Windows". Is there a newer version for download? Or how can I fix this issue?

Atlunch, 2018-11-27:
I have also the same issue when I'm trying to build 0.9.0.

dieteekanne2, 2018-11-29:
> 
> 
> Hi @dieteekanne2.
> Did you compiled the python API?
> 
> ```
> make PythonAPI
> ```

Hi @marcgpuig,
yes I did, still nothing changed

[85 more replies]