carla-simulator/carla #649: Packing on Win / Filenames too long

grafandreas, 2018-08-04:
Dear all,

having got the build finally running, UR editor opens Carla. I was trying to build a standalone by using "packaging->win64" and get errors while "cooking". Going through the logs I find:

CookResults: Error: Couldn't save package, filename is too long: D:/git/carla/carla/Unreal/CarlaUE4/Saved/Cooked/WindowsNoEditor/CarlaUE4/Content/Carla/Static/Pedestrians/Materials/Textures/ModelBased/femaleSkirtShirt/Female_Skirt_Shirt_SBoots_femaleDummy2_hair_mat_OcclusionRoughnessMetallic.uasset

A Windows Classic! Has anyone every packaged on WIndows 10? If so, what is your root level where you put git / carla etc.

Thanks,

Andreas

marcgpuig, 2018-08-20:
Hi @grafandreas,

It's not your computer problem. When packing, Unreal has a maximum file length of 260, that sometimes, [fails](https://answers.unrealengine.com/questions/60187/the-path-to-the-asset-is-too-long-for-cooking-im-l.html) anyway. My only suggestion is that you should work in `C:/carla/` for now.

@TheNihilisticRobot is working on it but renaming assets is not easy because it must be done inside Unreal Editor and looks like it must be done by hand.

Anyway, it's true that we have inconsistent and redundant paths/naming on certain files of our project like:
- Filenames ended in `OcclusionRoughnessMetallic.uasset` that can be changed to something like `ORM.uasset`. Also other image types like `_Normal` to `_n` and so on.
- Repeated filenames like:
  _see `Bl_House_SantaClaraSuburb009`_
   ```
  Bl_House_SantaClaraSuburb009_Bl_House_SantaClaraSuburb009_Frames_OcclusionRoughnessMetallic.uasset
   ```

- Redundant folders and filenames like:
  _see `FemaleSkinnyJeans` and `Female_SkinnyJeans`_
  ```
  FemaleSkinnyJeans/Female_SkinnyJeans_Tshirt_Boots_femaleDummy3_bottom_mat_OcclusionRoughnessMetallic.uasset
  ```
- Inconsistent id naming like:
  _see `9` and `009`_
  ```
  .../SntaClara_SuburbHouse_9/1024/Bl_House_SantaClaraSuburb009_Bl_House_SantaClaraSuburb009_Walls_OcclusionRoughnessMetallic.uasset
  ```
- Different naming styles (_CamelCase_, _pascalCase_, ...) like:
  _see `ModelBased`, `femaleDress`, `Female_Dress` and `shoes_mat`_
  ```
  .../Textures/ModelBased/femaleDress/Female_Dress_Shoes_femaleDummy1_shoes_mat_Normal.uasset
  ```

I'm sure at some point the artist team will rename everything correctly!

Cheers

nsubiron, 2018-08-20:
Blueprints/Vehicles/BmwGrandTourer/BmwGranTourer.uasset

Sakrust, 2018-10-08:
Are there any news concerning this issue? I am currently working directly on D:\ otherwise i'll get the same file name is too long error as mentioned above. But as we share a desktop among multiple users, this is not the best way to go.

marcgpuig, 2018-10-09:
Hey @Sakrust 
Which Carla version are you using? Our artists fixed it for 0.9.x some time ago, but we are pending to upload the assets because of other reasons :(

Sakrust, 2018-10-09:
Version 0.9.0, thanks!

marcgpuig, 2018-10-19:
@Sakrust Sorry for the late response, we are really busy these days :(
This will be fixed in 0.9.1 that will be available soon! :)

aTanguay, 2019-11-21:
Is this 256 character limit still true in 2019/11?

Mani2077, 2024-12-20:
> Is this 256 character limit still true in 2019/11?

its still true in 2024/12 :')