carla-simulator/carla #652: Can you provide a download list of `Content.tar.gz` for each version?

davidhopper2003, 2018-08-06:
I am in China. Due to well-known network reasons, I can't use the `Update.sh` file to download the dependent file `Content.tar.gz` in a terminal, but I can manually download it in a proxy tool. 
The [issue #131](https://github.com/carla-simulator/carla/issues/131) mentioned that we can manually download the dependent file, but it isn't my favorite version(I need version 0.9.0, but the provided one is version 0.7.0). Can you provide a download list of `Content.tar.gz` for each version so that I can pick the required version myself? thank you very much.

debnera, 2018-08-06:
The links are listed in the Util/ContentVersions.txt file:

> Here we keep a list of the Google Drive id's of the Content files. The last id
> in the list it's picked up by the Setup.sh script to download the last version
> of the contents.
> 
> You can direct download with https://drive.google.com/open?id=PUT_FILE_ID_HERE
> 
> 0.5.4: 0B2HFV-VuKn3PYUFFanlmZ2VMTW8
> 0.6.0: 1Gw8sw01hDEV4FtpHEZZwvS-6XN0jmaLT
> 0.7.0: 11PKC_lG7L80L1ZxiMlzV17utU45UcRnT
> 0.7.1: 1gf3YaFIjwr1EaK6qxq2V-a510Brt9Imq
> 0.8.0: 1tA-WwKA0RNhBoENYunqF3-F_wg11dL86
> 0.8.1: 1su5wQH7KJphAcirMA8VEDLndGigQ9tzq
> 0.8.2: 1llrBkZDvJmxXYh4r3jz0wuwz8Q8jRQeZ
> 0.8.3: 1z2XKOk9LoKI-EhxcrpI31W5TeYo2-jjn
> 0.8.4: 1mFNS-w5atQ45yegiTepRIeKBYVm8N8He
> 0.9.0: 1FtC00CrDb7Kz5StBAwb6vqOGbzZtpROx

For example, the link to 0.9.0 would be https://drive.google.com/open?id=1FtC00CrDb7Kz5StBAwb6vqOGbzZtpROx

davidhopper2003, 2018-08-06:
@debnera

closed 2018-08-07 by davidhopper2003

geekhut, 2019-04-17:
where do I put the Content.tar.gz file?

kunxianzheng, 2020-05-21:
For some reasons, many network proxies cannot be used, so I am looking for other ways to download the files in "carla-assets.s3.amazonaws.com". Then I found a link (https://bitbucket.org/carla-simulator/carla-content/src/master/), are the files in this link the official "Content files"? I mean, is this link official?

kunxianzheng, 2020-05-22:
> For some reasons, many network proxies cannot be used, so I am looking for other ways to download the files in "carla-assets.s3.amazonaws.com". Then I found a link [...]

I have found it, this link is indeed official, mentioned in "Get development assets" in https://carla.readthedocs.io/en/latest/build_update/#get-development-assets

Qdriving, 2020-11-17:
> I am in China. Due to the well-known network reason, I can't use the `Update.sh` file to download the dependent file `Content.tar.gz` in a terminal, but I can manually download it in a proxy tool.
> The [issue #131](https://github.com/carla-simulator/carla/issues/131) mentioned that we can manually download the dependent file, but it isn't my favorite version (I need version 0.9.0, but the provided one is version 0.7.0). Can you provide a download list of `Content.tar.gz` for each version so that I can pick the required one myself? Thank you very much.

Hello, 你要下载了最新的Content.tar.gz吗，能否共享一份到百度网盘，谢谢！

fatemehjdi, 2022-09-14:
Hi. for downloading the asset 0.5.4, it says that I need a permission. What can I do for that?

edugh02, 2024-03-14:
Hi, i am trying to download the 20211112_d5cfa12.tar.gz because i am doing a proyect with carla 0.9.13. The problem is that when I try to do the Update.bat within the carla repository, the following error appears:

Downloading "http://carla-assets.s3.amazonaws.com/20211112_d5cfa12.tar.gz"...
Excepción al llamar a "DownloadFile" con los argumentos "2": "Error en el servidor remoto: (404) No se encontró."
En línea: 1 Carácter: 1
+ (New-Object System.Net.WebClient).DownloadFile('http://carla-assets.s ...
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : WebException

 Exiting with error...

I tried to download with the link of Util/ContentVersions.txt file. In that file says that i have to search the link http://carla-assets.s3.amazonaws.com/20211112_d5cfa12.tar.gz, but the lin can be found. 
What i have to do?