carla-simulator/carla #653: some problem with the release of carla 0.9.0

meixialin, 2018-08-06:
I failed when I tried to run the new release .I follow the document and run the example script with python example.py , but I receive the response.
```
Traceback (most recent call last):
  File "example.py", line 114, in <module>
    main(add_a_camera=False, enable_autopilot=True)
  File "example.py", line 43, in main
    client = carla.Client('localhost', 2000)
AttributeError: 'module' object has no attribute 'Client'
```
Could you please tell me how to solve it ? Thank you !

StarryFire, 2018-08-12:
Did you find the solution for this? I have the same problem.

StarryFire, 2018-08-12:
Ok, solved it. 
You probably have anaconda installed on your system, so when you run 'python example.py' it uses the Anaconda environment python and not the one that is native to your system.
So, you will have to edit ~/.bashrc and comment the path to anaconda python

If you are using the compiled version of Carla, then all will work fine.
If you built Carla from the source, then will have to first run 'make PythonAPI', this will generate a .egg file in PythonAPI/dist/, now go into the example.py and edit the path to the .egg file accordingly
Now, the script should run fine on the source version too.

meixialin, 2018-08-13:
I didn't install anaconda,and I change the path to the .egg file as you say,but it didn't work.The result just as the same as before.

goksanisil23, 2018-08-15:
@StarryFire thanks for the solution. I can run the 'make PythonAPI', and it claims that it has successfully installed Carla lib in "C:/carla/PythonAPI\dist", but the dist folder actually doesnt exist. Have you built it on a Windows machine or Linux ?

wolfsubak, 2018-08-17:
I got same issue and there is only the 2.7 egg built in my dist folder.
The weird part: it was working well until I tried to use it with ros.

```
======================================================================
ERROR: test_default_values (test.test_transform.testLocation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/MyUsername/carla/PythonAPI/test/test_transform.py", line 14, in test_default_values
    location = carla.Location()
AttributeError: 'module' object has no attribute 'Location'
======================================================================
ERROR: test_client_version (test.test_client.testClient)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/wolf/MyUsername/carla/PythonAPI/test/test_client.py", line 17, in test_client_version
    c = carla.Client('localhost', 8080)
AttributeError: 'module' object has no attribute 'Client'

----------------------------------------------------------------------

```

pbaj, 2018-08-29:
I changed the path in example.py to the one as @StarryFire suggested and uninstalled anaconda. I yet get the same error. Can you please help with this. @meixialin were you able to solve the problem?

matthew-rococo, 2018-10-02:
i suggest you first "make clean" , and then "make launch" , and then "make PythonAPI", maybe this will correct your problem

mshaheerm, 2018-10-04:
[image]
I am trying to `make PythonAPI` from the main carla folder but it raises these build errors. how can i fix them?

iFuSiiOnzZ, 2018-10-04:
Check that `py` is available in you system. Open and command line and type `py`, it should open the default python version.

sunpochin, 2018-10-11:
I just encountered this issue, the cause might be I installed PythonClient from pre-compiled version 0.8.4 
using this:
"pip2 install -e .  --user --upgrade"
https://github.com/carla-simulator/carla/tree/master/carla_ros_bridge

After uninstall it  "pip uninstall carla" 
now my 
"python example.py" cmd won't find the incorrect version of carla.

gauravkumar22927, 2019-09-11:
I SOLVED IT ...
JUST INSTALL PYTHON 3.7 VERSION 64 BIT ONLY NOT 32 BIT IF YOUR COMPUTER IS 64 BIT 
this was cause of all error and install all packages "py -m pip install pkg_name" in **requirement.txt of all place**. requirement.txt will be at 2 places check it and one download is enough 2.9 gb dont go for other download.

geonational, 2019-12-16:
Thanks gauravkumar22927, I was trying to run it with python 3.8 64-bit, but is has to be 3.7 64-bit as you said.