carla-simulator/carla #658: UE4_ROOT is not defined, or points to a non-existant directory, please set this environment variable

lihanxiang0309, 2018-08-07:
Hello, 

When I run the "make launch", I got the following error: 

BuildCarlaUE4.sh: ERROR: UE4_ROOT is not defined, or points to a non-existant directory, please set this environment variable.

Previously I have successfully installed unRealEngine. Also, the $UE4_ROOT is actually defined:
echo $UE4_ROOT
/data/Files_Hanxiang/control-study/UnrealEngine_4.19

Any help will be appreciated.

Thanks,

Ethan

nsubiron, 2018-08-20:
Hi @lihanxiang0309,

Make sure that the folder actually exists. That error is raised if the folder does not exist.

closed 2018-10-11 by nsubiron

basavaraj1995, 2019-09-10:
`export UE4_ROOT=~/UnrealEngine_4.22`

SUNGBEOMCHOI, 2020-04-05:
Try it again
cd ~/UnrealEngine_4.24
./Setup.sh && ./GenerateProjectFiles.sh && make
cd ~/carla
./Update.sh
export UE4_ROOT=~/UnrealEngine_4.24
make launch

aruckman2, 2023-02-03:
Also having this issue on linux using Unreal 4.26

Kafuuaki, 2023-07-21:
Thank u!

IDontEatFish, 2023-07-21:
To be more precise with the issue:

1. Go to your directory of UnrealEngine
2. type the command ```pwd``` and remember the output
3. Set the variable UE4_ROOT=<the output of line 2> in either the shell that you call ```make launch``` in or in Carla/Util/BuildTools/BuildCarlaUE4.sh

UPTOLIMIT, 2023-10-19:
Add UE4_ROOT = ~/UnrealEngine-4.26 in the Vars.mk file

zhaosheng1105, 2025-02-13:
IF you try to add "export UE4_ROOT=~/UnrealEngine_4.22" in ~/.bashrc, and add  "UE4_ROOT=PATH.../UnrealEngine-4.26" in the first line of Carla/Util/BuildTools/BuildCarlaUE4.sh, BUT both are useless....

JUST PLZ try to reboot your computer!!