carla-simulator/carla #660: Help: there are no cars and pedestrians in the CARLA simulator.

davidhopper2003, 2018-08-08:
I have built Carla 0.9.0  from source code (based on Unreal Engine 4.19.2-release) and there are no errors during the building process.
I opened the simulator with the following command:
```
# Change to the directory of my Unreal Engine binaries.
cd ~/code/UnrealEngine/Engine/Binaries/Linux

# Open the UE4Editor with the CarlaUE4 project and Town01 map.
./UE4Editor ~/code/carla/Unreal/CarlaUE4/CarlaUE4.uproject /Game/Carla/Maps/Town01
```
The initial interface is shown below:
[image]

When I clicked the "Play" button in the simulator, there were no cars or pedestrians in the scene. 
Can you tell me What on earth the reason is. Thank you.
[image]

debnera, 2018-08-08:
I believe it is working as intended and currently you can only create cars (and sensors) through the Python API in 0.9.0. See the example Python files provided in the PythonAPI folder. 

The 0.9.0 is a very early version of the new client/networking API and it is still missing most of the previous features. Even pedestrians are not yet supported. You might be better of reverting to the 0.8.4 (or stable branch 0.8.2), unless you need the ability to control multiple vehicles through the Python API.

davidhopper2003, 2018-08-09:
@debnera 
I see. Thanks for your solution.

nsubiron, 2018-08-20:
Yes, this is the intended behaviour. At some point, we will add tools to spawn and control vehicles from the simulator itself, but for now, it's only possible to do so from the Python API.

closed 2018-10-11 by nsubiron