carla-simulator/carla #661: Is there a way to receive the torque applied to the wheels?

danielcomas25, 2018-08-08:
Hello,

I have been working a lot with Carla Simulator for the past weeks, and I find an excellent work and I really like it. I was implementing a longitudinal controller taking as input the throttle and the brake percentage, and taking the longitudinal speed as output. However, I wanted to go deeper on the controller and I am thinkig of using the torque applied to the wheels as the input. In the model of the car, there should be a function that converts the throttle into torque. Do you have any idea how can I translate to torque? Some people could say that I can calculate it with the acceleration of the car and the radio, the mass and the constant of inertia of the wheel. However, when a throttle pedal input is applied there is slip, so in this case I can not calculated from the state of the art equations. 

Thank you,

MrMushroom, 2019-02-13:
Hi!
I'd like to reopen this issue. As the ros-bridge got an update with a PID controller, it is necessary at some point to know the relation between brake force and torque applied to the wheels. Which results in more or less drastic braking.

For reference, the Dataspeed drive-by-wire kit for ros has relations between pedal_cmd_msgs and real life applied torque, as shown in the Figure below:
[image]
X-Axis is the brake cmd (float: 0 ... 1)
Y-Axis is the resulting torque

As one can see from the diagrams, the necessary breaking force (converted to torque) varies drastically. Therefore its very hard to get good PID controllers without knowing the related curve.

germanros1987, 2019-02-14:
This is officially re-opened. @manishthani could you bare this in mind for your current task?

manishthani, 2019-02-18:
Hi everyone,
Currently, I am working on the task of exposing physics parameters of the vehicles and wheels, so you can access and modify that data and use it according to your needs. I will keep in mind this issue when working on it so, please, let me know if you have more suggestions about it.
Thank you very much!

closed 2019-04-26 by stale[bot]

brscholz, 2020-11-12:
Hi everyone,

I have a quite similar issue, so I thought reopening this one might be better than creating a new one.
I'm trying to couple CARLA with a vehicle powertrain model in Simulink in the following way:

1) Accelerator and brake pedal position come from an external source and are fed into the Simulink model.
2) The Simulink model does the torque calculation from engine to wheel input.
3) CARLA receives the wheel torque and does the calculation of acceleration force and vehicle speed.
4) CARLA vehicle speed is fed into the Simulink model for calculation of the resulting engine speed.

Is there a way yet to directly set the torque that is applied to each wheel?

A workaround I thought about could be the actor.set_force API function, but that way I would lose effects of two- and four-wheel-drives and many more.

Thanks in advance and best regards
Robert

JulioBallinas, 2022-03-02:
Hi, same Issue, It would be fantastic for out team yo have a way to know the current torque apply to some object, or know the angular force of the object, don't know if there were something new, I couldn't find anything about it