carla-simulator/carla #667: Build issues with 0.9.0 and Ubuntu 16.04

fabianoboril, 2018-08-10:
Hello everyone,

I am trying to build Carla as described in the documentation on Ubuntu 16.04.
http://carla.readthedocs.io/en/latest/how_to_build_on_linux/
I am able to successfully download and build the UE4.19, but struggle with the Carla build.

My current issue is that Carla and rpclib-2.2.1 requires cmake 3.9, which is not part of Ubuntu 16.04
(btw. so is neither gcc-7). When I call "make launch", this is the output I receive:

```
Setup.sh: llvm-5.0 already installed.
Setup.sh: boost-1.67.0 already installed.
Setup.sh: Retrieving rpclib.
Cloning into 'rpclib-2.2.1-source'...
Note: checking out '3b00c4ccf480b9f9569b1d064e7a3b43585b8dfd'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Setup.sh: Building rpclib with libc++.
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.9.0 or higher is required.  You are running version 3.5.1

-- Configuring incomplete, errors occurred!
```

Is there a way to remain with cmake 3.5, and if not, is there a way to build cmake 3.9 on-the-fly, so as boost 1.67?

Thank you.

fabianoboril, 2018-08-13:
Meanwhile, I implemented a workaround.

Step 1: Replace cmake_minimum_required from 3.9 to 3.5
Step 2: Patch Setup.sh in Util/BuildTools
```
diff --git a/Util/BuildTools/Setup.sh b/Util/BuildTools/Setup.sh
index db0ca39..2ee0fee 100755
--- a/Util/BuildTools/Setup.sh
+++ b/Util/BuildTools/Setup.sh
@@ -137,6 +137,8 @@ else

   log "Building rpclib with libc++."

+  sed -i s/"3.9.0"/"3.5.0"/g ${RPCLIB_BASENAME}-source/CMakeLists.txt
+
   mkdir -p ${RPCLIB_BASENAME}-libcxx-build

   pushd ${RPCLIB_BASENAME}-libcxx-build >/dev/null
```
This did the trick, and obviously nowhere the features of CMake 3.9 are used.

Best regards,
Fabain

nsubiron, 2018-08-20:
Hi @fabianoboril,

Thanks for the report, maybe this can be reported @rpclib too.

Meanwhile I will add to the documentation that you should have installed CMake 3.9 or higher.

commit 2bf6d122 (2018-08-20): Add cmake version to documentation, close #667
closed 2018-08-20 by nsubiron