carla-simulator/carla #671: Available vehicle library and Default vehicle

n-akram, 2018-08-12:
Hello everyone,

I am using CARLA 0.8.4 and I am trying to run it by changing default vehicle to two wheeled vehicle, but I am not able to.

I have seen that I can change the model in initial settings at, 'PlayerVehicle', like this
"PlayerVehicle=/Game/Blueprints/Vehicles/BmwGrandTourer/BmwGranTourer.BmwGranTourer_C"

But my problem is, I am not able to find these blueprints (Since I have not built CARLA from source, but just using pre-compiled versions). Can someone help me in finding models for bike?

Note: As it is easy enough to change the default vehicle by knowing the models and path, I think it would be good idea to have a text file with all the vehicle models available. It will make it easier for the documentation.

PS: Please excuse me if this question is really basic or if there is already such file existing. I have not been in touch with CARLA from past few months.

n-akram, 2018-08-29:
@marcgpuig 

Thank you for your reply. It helps a lot. 

And I hope you had a nice time on vacation. :-)

marcgpuig, 2018-08-30:
Let me reopen this issue until the spelling errors got fixed :)

closed 2018-11-05 by stale[bot]