carla-simulator/carla #676: how to load other maps and obtain 2d map data?

WillenPeng, 2018-08-14:
Hi,

I run carla 0.8.2 with ubuntu. I want to ask how to load other maps such as Town02.umap and testmaptruck.umap in Unreal Engine. I had take a look at the tutorials, but not found /Game/carla/Maps. I also run the maps I found in Unreal Engine folder, but it still not work.

Another question how to obtain the corresponding 2d map? can i use ros to publish this info as a topic. Because I want to use this info to implement other algorithms.

Thanks

hxlee309, 2018-08-16:
We saved all the maps in the folder called "/Games/Maps/". Currently we have two big problems:

1. The only maps we can load are Town01.umap and Town02.umap
2. After significantly modifying the map Town01.umap using editor, and run the simulator using the command "./CarlaUE4.sh /Game/Maps/Town01.umap", what we can see is still the old Town01 map instead of the modified one.

Any help will be appreciated.

Hanxiang

danil-tolkachev, 2018-08-16:
Hi @WillenPeng 

For the second question look at the #537 (it's not merged yet). I used `CarlaMap` class to generate ROS OccupancyGrid message.

lihanxiang0309, 2018-08-17:
Hi @danil-tolkachev 

Thanks it works! Now we succeed in generating the 2d occupancy map (2600 x 2200). However, we are confused about how to convert the world x-y coordinate into the 2d map pixel index?

Thanks

danil-tolkachev, 2018-08-20:
There is a `convert_to_pixel` method in the `CarlaMap` class. You can also look at the `view_start_positions.py` example.

chengjinjack, 2018-09-20:
@hxlee309 
hello!
Excuse me, have you solved the second question?

closed 2018-11-26 by stale[bot]