carla-simulator/carla #678: How to run build-from-source simulator

hxlee309, 2018-08-16:
Hi,

I successfully built carla 0.8.2 from source. However, the official tutorial only tells us how to launch the editor, but didn't tells us how to run the simulator. Please tell me how to run it.

Thanks,

Hanxiang

closed 2018-10-19 by marcgpuig

yeduzhouzi, 2019-08-18:
Hi @marcgpuig,

for some of the Linux distros (like the one I use) the only option to get Nvidia graphic card work is to use the bumblebee solution, with which I have to explicitly add "optirun" in command line to force the Nvidia graphic card to be started. I could get the Unreal editor launched in this way with Nvidia graphic card activated, but then when I hit "play" within the editor and try to start CARLA, it won't be launched by Nvidia graphic card. Do you have any idea how I can start CARLA from command line under this circumstance? Or do you have any other way out?

marcgpuig, 2019-08-26:
Hi @yeduzhouzi.
Maybe you can try to make a package and specifically run the package using the `optirun` command.

qhaas, 2020-01-06:
> Hello @hxlee309,
> 
> Please read the [F.A.Q](http://carla.readthedocs.io/en/latest/faq/):
> 
> > Once you open the project in the Unreal Editor, you can hit Play to test CARLA.
> 
> Also, take a look at the issue #22.

@marcgpuig Is there a command to issue this 'hit play' behavior in a headless / command-line-only environment (e.g. command-line argument or INI config file change)?  I've built CARLA 0.9.7 from source and it is running via the UE4Editor per `nvidia-smi`, but `lsof` shows that only port 1985 is being listened to by UE4Editor.  A client using the PythonAPI isn't able to connect to whatever is listening on 1985.  The logs and documentation indicate that port 2000 should be the port that a client connects to, but no process is listening to that port.

Thanks!