carla-simulator/carla #683: Simulator and ROS Returns Different Player Position

hxlee309, 2018-08-20:
Hi,

[image]
[image]

The first image attached is about the player info we get from the simulator, and the second image attached is about the player info we get from a ROS topic called /player_vehicle. We can see that the sign of y-position is different. What might go wrong in this case?

Thank,

Hanxiang

danil-tolkachev, 2018-08-20:
As far as I know this is due to the fact that the Carla uses left handed coordinate system, and ROS uses right handed one.

closed 2018-10-27 by stale[bot]

frk1993, 2020-07-05:
Hi,

Is the coordinate system in the newest version also left-handed?

Thanks!

werewolfdev, 2021-10-15:
@frk1993 yes CARLA still uses the UE4 left handed coordinate system