carla-simulator/carla #684: Player Vehicle and Non-player Agents Not visible

ptiwari0664, 2018-08-21:
Hello all,

I am a newbie in carla simulator. I have build carla from the source and also able to make binary out of it. 

**Problem:** While executing the command "make launch", Unreal Engine's Editor launches and I am able to see town01 in it. However, player vehicle and other cars and pedestrians are not visible.

Does anyone has a solution of it ??
I will be glad to receive any help. Thanks in advance.

ptiwari0664, 2018-08-21:
hi Marc,

Thanks for replying. I checked the reference which you gave. 
Yes, I am using 0.9.x version, however it seems, this version is not going to work for me as the ROS support is not there. 

Thanks.

closed 2018-08-21 by marcgpuig

christianjcc, 2020-10-01:
Anyone know what the default CARLA driver model (e.g. car-following, lane-change modeling)  is based on? Are the non-player agents/vehicles based on say the Intelligent Driver Model (IDM) or Optimal Velocity Model (OVM) for car-following model? Could not track down any document explicitly describing what the AI non-player vehicles model is modeled after. Any information that can be provided is greatly appreciated.